instrumentkit/InstrumentKit #52: Support for Prologix GPIB-USB and GPIB-Ethernet Adapters

pumphaus, 2015-10-28:
Support for the [Prologix GPIB USB](http://prologix.biz/gpib-usb-controller.html)/[Ethernet Adapters ](http://prologix.biz/gpib-ethernet-controller.html) would be nice. I'm currently working on an implementation, but I'm unsure as to how to plug it into the existing API:
The method ``open_gpibusb`` hardcodes the Galvant USB Adapter. On Linux (and possibly OS X) some lookups in /sys/class/tty can reveal the manufacturer of the adapter and intelligently choose the correct backend, but I'm unsure if there exists a similar facility on Windows.
Worst case, a keyword argument or an entirely seperate method has to be added, I guess.

scasagrande, 2015-10-29:
We could do something where an optional parameter is given to select the brand. Thankfully the protocol is very similar between the two. The main difference is the serial port settings. We could add another optional parameter to `serial_manager.new_serial_connection` for the flow control, and then modify `Instrument.open_gpibusb` to expose the baud and flow control parameters.

glennlive, 2016-04-15:
What is the state of support for the Proligix devices?  Is there a development branch available publicly?

scasagrande, 2016-04-16:
Development focus has been on many items lately, but Prologix adapters has not been on that list.

pumphaus, 2016-04-16:
Since our project is built upon Python3 and back then InstrumentKit only supported Python2, I had stopped working on the Prologix support.
Now that Python3 support is there, I can look into it again. I don't know when I'll get to that, however.

glennlive, 2016-04-17:
Did you have anything worth looking at, as I am considering using instrumentKit in my lab, but would need to add support for prologix, at least in a minimalistic fashion.

scasagrande, 2016-04-17:
I'm curious, do the Prologix units actually use proper GPIB line drivers yet?

glennlive, 2016-04-18:
The unit I have has been laying around the lab for >5 years now, so I can't speak to their new design having proper GPIB line drivers or not.  The one I have has the mega644 connected directly up to the 24-pin connector, without so much as a series resistor...  

That said, maybe the better question would, what is the preferred Ethernet based GPIB adapter to use with InstrumentKit?

scasagrande, 2016-04-18:
@cgranade and I came up with the idea a few years ago to simply have a pass-through script running on a RPi. I still like that idea as it lets people "upgrade" their gpibusb or serialusb adapters to ethernet.

cgranade, 2016-04-19:
Though I've not done so myself, it seems as though ``socat`` can be used to host serial connections via TCP. For security reasons I'd recommend tunneling such through SSH, to be sure.

scasagrande, 2022-01-28:
Oh, this was added back in #221

closed 2022-01-28 by scasagrande