instrumentkit/InstrumentKit #58: Switch from nose to pytest

scasagrande, 2016-01-14:
It has come to my attention that the test framework `nose` (including `nose2`) is not being as actively developed as `py.test`. It has a lot more features and nicer output. It also claims to support nose-tests straight out of the box. I believe the only thing we will need to do is replace any `nose.eq_` calls with just straight `assert` statements, as well as update travis & docs.

https://github.com/pytest-dev/pytest/
http://pytest.org/latest/

Acceptance Criteria
-------------------------
- Tests run and pass using the pytest framework 
- Travis updated
- Dev docs updated
- dev-requirements.txt updated

bilderbuchi, 2017-07-14:
OK, so I took a first stab at this (on Python 3.6). 
* I fixed the `mock import` occurences to import mock from instruments.tests, where I have placed an import guard to choose the stdlib mock that's available in Python>=3.3
* I realised that the dependency on pyserial is actually pyserial>=3.3, since you are using `list_ports_common`.
* With those changes, and nose installed, `pytest` runs the same 675 tests, and succeeds, as `nosetests`. So, as long as nose is installed, people can already use pytest
* Nose functionality used is `@raises`, `eq_` and `nottest` (once). `eq_`
  * I think is easy to replace with a nice regular expression search/replace: basically, `eq_(a,b)` to `assert a == b`.
  * nottest occurs only once, so probably not much effort to replace.
  * `@raises` decorator is used 161 times, and is more tedious to replace, since pytest uses this with a [context manager](http://pytest.readthedocs.io/en/reorganize-docs/new-docs/user/pytest_raises.html).

scasagrande, 2017-07-14:
Neat, thanks for taking a look.

I forgot about the pyserial minimum version requirement. There is some more stuff in PR that uses newer features of pyserial, so I'll have to think about what to do there.

bilderbuchi, 2017-07-15:
Optimally, we should find out how much effort it is to create a conda-forge recipe for pyserial, then the min version won't be a problem. I think the pymeasure folks have tried this before, correct @cjermain?

bilderbuchi, 2017-08-13:
update: Recently, a current version of pyserial has become available on conda-forge, so that can be installed via conda now.

bilderbuchi, 2018-03-16:
So, this was a lot of manual work, but I have created a PR (#183).

PR #183 (2018-03-24): Nose to pytest transition
Move from nose to pytest. Fixes #58.

The commit messages have some explanations, basically this was more or less tedious copy/paste. A lot of code was touched because the transition to `pytest.raises` necessitates increasing indentation for a whole block. (You can inspect changes without considering whitespace by appending `?w=1` to the URL)

Pytest tests all pass, the same number as before with nose. Let's see what the CI says.
closed 2018-03-24 by scasagrande