instrumentkit/InstrumentKit #88: setup.py install script does not work with python3

CatherineH, 2016-03-03:
I'm running:

```
python3 setup.py install
```
and I get the error:

```
Traceback (most recent call last):
  File "setup.py", line 11, in <module>
    import instruments as ik
  File "/home/catherine/python_projects/InstrumentKit/instruments/instruments/__init__.py", line 11, in <module>
    from . import abstract_instruments
  File "/home/catherine/python_projects/InstrumentKit/instruments/instruments/abstract_instruments/__init__.py", line 6, in <module>
    from instruments.abstract_instruments.instrument import Instrument
  File "/home/catherine/python_projects/InstrumentKit/instruments/instruments/abstract_instruments/instrument.py", line 32, in <module>
    import urlparse
ImportError: No module named 'urlparse'
```

I think urlparse is no longer available with python3

scasagrande, 2016-03-04:
Yeah, `urlparse` has been moved to 
```python
from urllib.parse import urlparse
```
Where the easiest solution is:
```python
from future.standard_library import install_aliases
install_aliases()
from urllib.parse import urlparse
```
Which will make it work on 2&3

Which once I'm done linting I'm going to look to start to see how much work is needed for py3 support. This one is first on my list, then hopefully I can start running the tests.

cgranade, 2016-03-04:
Is the ``future`` library already a dependency?

scasagrande, 2016-03-04:
Yep! It was needed for some of the other Py3 work that I did earlier.

cgranade, 2016-03-04:
Ah, cool! I missed that, then, sorry.

scasagrande, 2016-03-13:
Solved with PR #93

closed 2016-03-13 by scasagrande