instrumentkit/InstrumentKit #104: Error Python 3.5

ricardorector, 2016-04-22:
import instruments as ik

scope = ik.tektronix.TekTDS224.open_gpibusb('/dev/tty.usbserial-DJ000GVP',1)

print (scope.name)

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-3-3b16a47807e0> in <module>()
----> 1 print (scope.name)

/Applications/MAMP/htdocs/src/instrumentkit/instruments/generic_scpi/scpi_instrument.py in name(self)
     51         :rtype: `str`
     52         """
---> 53         return self.query('*IDN?')
     54 
     55     @property

/Applications/MAMP/htdocs/src/instrumentkit/instruments/abstract_instruments/instrument.py in query(self, cmd, size)
    132             value = self.read(size)
    133         else:
--> 134             value = self._file.query(cmd, size)
    135         if self.prompt is not None:
    136             prompt = self.read(len(self.prompt))

/Applications/MAMP/htdocs/src/instrumentkit/instruments/abstract_instruments/comm/abstract_comm.py in query(self, msg, size)
    231         if self.debug:
    232             self._logger.debug(" <- %s", repr(msg))
--> 233         resp = self._query(msg, size)
    234         if self.debug:
    235             self._logger.debug(" -> %s", repr(resp))

/Applications/MAMP/htdocs/src/instrumentkit/instruments/abstract_instruments/comm/gi_gpib_communicator.py in _query(self, msg, size)
    339         :rtype: `str`
    340         """
--> 341         self.sendcmd(msg)
    342         if '?' not in msg:
    343             self._file.sendcmd('+read')

/Applications/MAMP/htdocs/src/instrumentkit/instruments/abstract_instruments/comm/abstract_comm.py in sendcmd(self, msg)
    216         if self.debug:
    217             self._logger.debug(" <- %s", repr(msg))
--> 218         self._sendcmd(msg)
    219 
    220     def query(self, msg, size=-1):

/Applications/MAMP/htdocs/src/instrumentkit/instruments/abstract_instruments/comm/gi_gpib_communicator.py in _sendcmd(self, msg)
    317         self.timeout = self.timeout
    318         time.sleep(sleep_time)
--> 319         self.eos = self.eos
    320         time.sleep(sleep_time)
    321         self._file.sendcmd(msg)

/Applications/MAMP/htdocs/src/instrumentkit/instruments/abstract_instruments/comm/gi_gpib_communicator.py in eos(self, newval)
    226                 newval = 3
    227             else:
--> 228                 raise ValueError("EOS must be CRLF, CR, LF, or None")
    229             self._file.sendcmd("++eos {}".format(newval))
    230 

ValueError: EOS must be CRLF, CR, LF, or None

scasagrande, 2016-04-23:
WHOOPS I'll fix that asap

scasagrande, 2016-04-23:
Please see PR #106

ricardorector, 2016-04-23:
Ok here is the result.

OSX.

scope = ik.tektronix.TekTDS224.open_gpibusb('/dev/tty.usbserial-DJ000GVP',1) OK
print (scope.name) OK
R = 
TEKTRONIX,TDS 220,0,CF:91.1CT FV:v2.03 TDS2CM:CMV:v1.04
print (scope.query('*IDN?'))
R = TEKTRONIX,TDS 220,0,CF:91.1CT FV:v2.03 TDS2CM:CMV:v1.04

[x,y] = scope.channel[0].read_waveform()

---------------------------------------------------------------------------
UnicodeDecodeError                        Traceback (most recent call last)
<ipython-input-5-91eb23dc50f8> in <module>()
----> 1 [x,y] = scope.channel[0].read_waveform()

/Applications/MAMP/htdocs/src/instrumentkit/instruments/tektronix/tektds224.pyc in read_waveform(self, bin_format)
     90 
     91             yoffs = self._tek.query(
---> 92                 'WFMP:{}:YOF?'.format(self.name))  # Retrieve Y offset
     93             ymult = self._tek.query(
     94                 'WFMP:{}:YMU?'.format(self.name))  # Retrieve Y multiply

/Applications/MAMP/htdocs/src/instrumentkit/instruments/abstract_instruments/instrument.pyc in query(self, cmd, size)
    132             value = self.read(size)
    133         else:
--> 134             value = self._file.query(cmd, size)
    135         if self.prompt is not None:
    136             prompt = self.read(len(self.prompt))

/Applications/MAMP/htdocs/src/instrumentkit/instruments/abstract_instruments/comm/abstract_comm.pyc in query(self, msg, size)
    231         if self.debug:
    232             self._logger.debug(" <- %s", repr(msg))
--> 233         resp = self._query(msg, size)
    234         if self.debug:
    235             self._logger.debug(" -> %s", repr(resp))

/Applications/MAMP/htdocs/src/instrumentkit/instruments/abstract_instruments/comm/gi_gpib_communicator.pyc in _query(self, msg, size)
    342         if '?' not in msg:
    343             self._file.sendcmd('+read')
--> 344         return self._file.read(size).strip()

/Applications/MAMP/htdocs/src/instrumentkit/instruments/abstract_instruments/comm/abstract_comm.pyc in read(self, size, encoding)
    203         :rtype: `str`
    204         """
--> 205         return self.read_raw(size).decode(encoding)
    206 
    207     def sendcmd(self, msg):

/Users/Rich/anaconda/lib/python2.7/site-packages/future/types/newbytes.pyc in decode(self, encoding, errors)
    207             register_surrogateescape()
    208 
--> 209         return newstr(super(newbytes, self).decode(encoding, errors))
    210 
    211         # This is currently broken:

/Users/Rich/anaconda/lib/python2.7/encodings/utf_8.pyc in decode(input, errors)
     14 
     15 def decode(input, errors='strict'):
---> 16     return codecs.utf_8_decode(input, errors, True)
     17 
     18 class IncrementalEncoder(codecs.IncrementalEncoder):

UnicodeDecodeError: 'utf8' codec can't decode byte 0xe1 in position 44: invalid continuation byte

scasagrande, 2016-04-23:
Are you using Python 3.5 or 2.7? I'm seeing references to 2.7 in your debug messages

ricardorector, 2016-04-23:
jajaja Sorry change at 2.7 and i dont remember.

scasagrande, 2016-04-23:
Ok, I have a TekTDS224 so I'm going to try to reproduce your error

scasagrande, 2016-04-23:
Figured it out, the connection channel needed a larger timeout value for when transferring waveforms.

closed 2016-04-24 by scasagrande