instrumentkit/InstrumentKit #105: Error Python 3.5

ricardorector, 2016-04-23:
On windows.

import instruments as ik

scope = ik.tektronix.TekTDS224.open_serial("COM3",460800)

print (scope.name)
Res: TEKTRONIX,TDS 220,0,CF:91.1CT FV:v2.03 TDS2CM:CMV:v1.04

[x, y] = scope.channel[0].read_waveform()

---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
<ipython-input-32-229930d78b5d> in <module>()
----> 1 [x, y] = scope.channel[0].read_waveform()

c:\spb_data\src\instrumentkit\instruments\tektronix\tektds224.py in read_waveform(self, bin_format)
     83                 self._tek.sendcmd('CURVE?')
     84                 raw = self._tek.binblockread(
---> 85                     data_width)  # Read in the binary block,
     86                                                     # data width of 2 bytes
     87 

c:\spb_data\src\instrumentkit\instruments\abstract_instruments\instrument.py in binblockread(self, data_width, fmt)
    267         symbol = self._file.read(1)
    268         if symbol != "#":  # Check to make sure block is valid
--> 269             raise IOError("Not a valid binary block start. Binary blocks "
    270                           "require the first character to be #.")
    271         else:

OSError: Not a valid binary block start. Binary blocks require the first character to be #.

ricardorector, 2016-04-23:
What happen? Im using Anaconda Python 3, Jupyter Notebook before Python Notebook.
And has a lot of issues???

scasagrande, 2016-04-23:
I'm not surprised there are still Py3 issues, you're the first to really be using it! We don't yet have any test coverage for the `binblockread` function, so yep, I'm not surprised its not working. I know whats wrong though, so should be an easy fix (along with the other issue you reported, thank you).

Please keep pasting any additional issues you find and I will fix them in the morning.

scasagrande, 2016-04-23:
Wait, why are you using `open_serial` with a baud of `460800` on the `TekTDS224`? I'm pretty sure that that device can't handle a direct serial connection with a baud rate that high.

ricardorector, 2016-04-23:
I try with 115200 and not connect.

ricardorector, 2016-04-23:
Using Windows.
Ok in open_serial('COM3',115200)
Works.

Using Windows
open_gpibusb('COM3',1)

print (scope.name);
Value Error the same of issue #104

Using Linux and Mac.
Same issue #104 
Using your GPIB Adapter Rev. 4

scasagrande, 2016-04-23:
Ok, fix for #104 incoming

closed 2016-04-24 by scasagrande