instrumentkit/InstrumentKit #108: requirements problem from PyPI

CatherineH, 2016-04-25:
I just tried installing InstrumentKit from pip, and I get this dependency problem:

```
Downloading/unpacking pyusb (from InstrumentKit)
  Could not find a version that satisfies the requirement pyusb (from InstrumentKit) (from versions: 1.0.0b1, 1.0.0b2, 1.0.0a3, 1.0.0rc1, 1.0.0a2, 1.0.0a3, 1.0.0a2)
```

Any ideas?

scasagrande, 2016-04-25:
Python version?

CatherineH, 2016-04-25:
3.4

scasagrande, 2016-04-25:
I made new virtualenvs of both py2.7 and py3.5 and installed with the pip cache disabled, everything went fine. Maybe this is a Windows issue (assuming you're on your lab computer)?

CatherineH, 2016-04-25:
It happened on both windows and linux...

scasagrande, 2016-04-25:
Maybe you have some weird stuff cached in pip, make sure you don't have stuff already installed with `pip list`

Also try `pip install --no-cache-dir instrumentkit`

scasagrande, 2016-04-25:
(I also just tried a new virtualenv with python 3.4, no problems)

CatherineH, 2016-04-25:
hmm, okay, that worked.

closed 2016-04-25 by CatherineH

scasagrande, 2016-04-25:
[image]