instrumentkit/InstrumentKit #110: pyvisa version check broken

thevorpalblade, 2016-04-30:
abstract_instruments/comm/visa_communicator.py on lines 49 and 50
as well as
abstract_instruments/instrument.py on line 528
have pyvisa version checks that go like:
`if int(visa.__version__.replace(".", "")) >= 160:`
or similar.
With pyvisa 1.8
`visa.__version__.replace(".", "")`
evaluates to 18, which causes the code to fail the version check and use the code for the old version of pyvisa rather than the new version.

For my installation of pyvisa 1.8, changing 160 to 16 everywhere ot appears fixes the problem, but that might break with older versions of pyvisa?

scasagrande, 2016-05-01:
Hmmm, interesting. I did not know that, let me think about the best solution here.

scasagrande, 2016-05-01:
I could have sworn that older versions of `pyvisa` required some sort of VISA lib to be installed, but it doesn't seem like that is the case anymore. I'll explore making `pyvisa` a non-optional requirement since its a small pure Python library.

thevorpalblade, 2016-05-01:
I guess I should clarify!
pyvisa had an API change a while ago, version 1.6 being the first version for the new API.
So, the version check I refer to above checks which version of pyvisa is running and uses the old or new API accordingly.

For the current version of pyvisa (1.8, the one I happened to have installed) the version check 
incorrectly identifies the pyvisa version as the _old_ API, and promptly breaks. So, the thing that needs to be fixed is the version checking code.

As to your other comment, indeed, pyvisa requires the official National Instruments VISA libraries to be installed. I think an optional dependency makes lots of sense. You might consider dropping support for old versions of pyvisa though, as they are deprecated.

scasagrande, 2016-05-01:
Yeah, I remember them changing their API in 1.6. For now I'll just do a fix to the version checking while I decide what to do.

And yes, in order for pyvisa to actually function it requires VISA drivers to be installed, but what I was referencing is how pyvisa doesn't seem to error out during install if VISA is missing.

scasagrande, 2016-05-01:
Should be fixed now.

closed 2016-05-01 by scasagrande