instrumentkit/InstrumentKit #120: NotImplementedError on opening a usbtmc device

CatherineH, 2016-06-08:
On the following code:

```
import instruments as ik

port = "USB::0x0699::0x0365::INSTR"
tek = ik.tektronix.TekTDS224.open_usbtmc(port)
```

I'm getting the following error:
```
Traceback (most recent call last):
  File "~/InstrumentKit/doc/examples/ex_oscilloscope_gui.py", line 5, in <module>
    tek = ik.tektronix.TekTDS224.open_usbtmc(port)
  File "~/InstrumentKit/instruments/abstract_instruments/instrument.py", line 580, in open_usbtmc
    return cls(usbtmc_comm)
  File "~/InstrumentKit/instruments/tektronix/tektds224.py", line 164, in __init__
    self._file.timeout = 3 * pq.second
  File "~/InstrumentKit/instruments/abstract_instruments/comm/usbtmc_communicator.py", line 73, in timeout
    raise NotImplementedError
NotImplementedError
```

CatherineH, 2016-06-08:
For now in my code, I've replaced the timeout properties of USBTMCCommunicator with:

```
    @property
    def timeout(self):
        return self._filelike.timeout

    @timeout.setter
    def timeout(self, newval):
        self._filelike.timeout = newval
```

scasagrande, 2016-06-09:
Ah yes, drr.

closed 2016-06-24 by scasagrande