instrumentkit/InstrumentKit #127: libusb0-dll:err [control_msg] on usbtmc

CatherineH, 2016-06-14:
I've moved the tektronix scope back to the lab, so I need to test the latest bug fixes on the windows lab computer instead of my linux computer. I've encountered an error generated by libusb/pyusb, however, it doesn't occur with usbtmc alone. For comparison, here is the working code in usbtmc:

```
from usbtmc import Instrument
instr = Instrument(0x0699, 0x0365)
print(instr.ask("*IDN?"))
```
This results in the expected result.

and here is my equivalent InstrumentKit code:

```
from instruments.tektronix import TekTDS224
tek = TekTDS224.open_usbtmc(idVendor=0x0699, idProduct=0x0365)
print(tek.query("*IDN?"))
```

which produces the error:

```
Traceback (most recent call last):
  File "~\InstrumentKit\doc\examples\tektronix\usbtmc_example.py", line 8, in <module>
    print(tek.query("*IDN?"))
  File "~\InstrumentKit\instruments\abstract_instruments\instrument.py", line 132, in query
    value = self._file.query(cmd, size)
  File "~\InstrumentKit\instruments\abstract_instruments\comm\abstract_comm.py", line 233, in query
    resp = self._query(msg, size)
  File "~\InstrumentKit\instruments\abstract_instruments\comm\usbtmc_communicator.py", line 181, in _query
    return self._filelike.ask(msg, num=size, encoding="utf-8")
  File "C:\Python34\lib\site-packages\python_usbtmc-0.7-py3.4.egg\usbtmc\usbtmc.py", line 575, in ask
  File "C:\Python34\lib\site-packages\python_usbtmc-0.7-py3.4.egg\usbtmc\usbtmc.py", line 555, in write
  File "C:\Python34\lib\site-packages\python_usbtmc-0.7-py3.4.egg\usbtmc\usbtmc.py", line 464, in write_raw
  File "C:\Python34\lib\site-packages\python_usbtmc-0.7-py3.4.egg\usbtmc\usbtmc.py", line 338, in open
  File "C:\Python34\lib\site-packages\python_usbtmc-0.7-py3.4.egg\usbtmc\usbtmc.py", line 645, in clear
  File "C:\Python34\lib\site-packages\pyusb-1.0.0rc1-py3.4.egg\usb\core.py", line 1043, in ctrl_transfer
  File "C:\Python34\lib\site-packages\pyusb-1.0.0rc1-py3.4.egg\usb\backend\libusb0.py", line 588, in ctrl_transfer
  File "C:\Python34\lib\site-packages\pyusb-1.0.0rc1-py3.4.egg\usb\backend\libusb0.py", line 425, in _check
usb.core.USBError: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed, win error: The I/O operation has been aborted because of either a thread exit or an application request.\r\n\n'
```

This bug is probably difficult to diagnose from a distance without a usbtmc device, so I'll poke around.

scasagrande, 2016-06-14:
This is only on Windows?

scasagrande, 2016-06-14:
So the underlying call to `usbtmc` package looks like this:

```python
return self._filelike.ask(msg, num=size, encoding="utf-8")
```
Where `size=-1`

So my initial reaction is that it doesn't like the extra keywords. Or if this is Windows-specific, maybe there is some sort of port-related problem going on (open, close, conflicts, etc).

CatherineH, 2016-06-14:
So far it has only occurred on Windows

scasagrande, 2016-06-15:
Hmm, could you try this on Windows:

```python
from usbtmc import Instrument
instr = Instrument(0x0699, 0x0365)
print(instr.ask("*IDN?", size=-1, encoding="utf-8"))
```

CatherineH, 2016-06-15:
It still works.

CatherineH, 2016-06-15:
Digging further...

I placed .ask() commands at several steps, one within the USBTMCCommunicator initialization, one  within open_usbtmc, and the final after open_usbtmc(). 

The first two execute as expected, but the final results in a timeout error:

```
in initialization  TEKTRONIX,TDS 2004B,C033931,CF:91.1CT FV:v22.01
in open_usbtmc,  TEKTRONIX,TDS 2004B,C033931,CF:91.1CT FV:v22.01
Traceback (most recent call last):
  File "~\InstrumentKit\doc\examples\tektronix\usbtmc_example.py", line 14, in <module>
    print("in main, ", tek._file._filelike.ask("*IDN?"))
  File "C:\Python34\lib\site-packages\python_usbtmc-0.7-py3.4.egg\usbtmc\usbtmc.py", line 576, in ask
  File "C:\Python34\lib\site-packages\python_usbtmc-0.7-py3.4.egg\usbtmc\usbtmc.py", line 559, in read
  File "C:\Python34\lib\site-packages\python_usbtmc-0.7-py3.4.egg\usbtmc\usbtmc.py", line 508, in read_raw
  File "C:\Python34\lib\site-packages\pyusb-1.0.0rc1-py3.4.egg\usb\core.py", line 402, in read
  File "C:\Python34\lib\site-packages\pyusb-1.0.0rc1-py3.4.egg\usb\core.py", line 988, in read
  File "C:\Python34\lib\site-packages\pyusb-1.0.0rc1-py3.4.egg\usb\backend\libusb0.py", line 537, in bulk_read
  File "C:\Python34\lib\site-packages\pyusb-1.0.0rc1-py3.4.egg\usb\backend\libusb0.py", line 622, in __read
  File "C:\Python34\lib\site-packages\pyusb-1.0.0rc1-py3.4.egg\usb\backend\libusb0.py", line 425, in _check
usb.core.USBError: [Errno None] b'libusb0-dll:err [_usb_reap_async] timeout error\n'
```

CatherineH, 2016-06-15:
gah! how did this line:

```
self._file.timeout = 3 * pq.second
```

not get taken out of the develop branch?

CatherineH, 2016-06-15:
Commenting out the line setting the timeout resolved that error. I removed this line in my PR. 

I should have been more explicit in my PR about what needed to be changed in order to fix the tektronix issues.

scasagrande, 2016-06-15:
I can't just remove that line though, because its needed for the actual TekTDS224 when transferring waveforms over serial/gpibusb because they take longer than the common 1sec timeout.

I took a look at the source of the `usbtmc` package, and it seems to accept a value in seconds, and then multiplies it by 1000. Maybe that's actually suppose to be milliseconds, and then it gets converted to microseconds? If so that's a simple fix, I'd just change

```python
newval = assume_units(newval, pq.second).rescale(pq.second).magnitude
```
to
```python
newval = assume_units(newval, pq.second).rescale(pq.millisecond).magnitude
```

scasagrande, 2016-06-16:
Here is the relevant line in `python-usbtmc` that I'm interacting with:

https://github.com/python-ivi/python-usbtmc/blob/master/usbtmc/usbtmc.py#L198

scasagrande, 2016-06-16:
I wonder if this has anything to do with libusb0, maybe it doesn't like timeouts greater than 1?

scasagrande, 2016-06-16:
On my linux install on my laptop, I have both libusb major versions installed (both `libusb-0.1-4` and `libusb-1.0-0`), so I wonder if it is working on Linux for you because its accessing libusb1 vs trying to use libusb0 on your Windows box.

scasagrande, 2016-06-16:
I also noticed in one of your commits you did this in the `TekTDS224` class:

```python
self._file.timeout = 1000
```

And that works for usbtmc? O_o

CatherineH, 2016-06-23:
confirmed: switching to:

```
newval = assume_units(newval, pq.second).rescale(pq.millisecond).magnitude
```

fixes the problem

CatherineH, 2016-06-23:
I've committed the change to the develop branch.

closed 2016-06-23 by CatherineH