instrumentkit/InstrumentKit #160: Shebang Lines In Wrong Places

MichalKononenko, 2017-03-07:
If I understand correctly, the purpose of the ``#!`` line is to indicate that a particular python file is executable. However, these lines are scattered around library code, in places like [this](https://github.com/Galvant/InstrumentKit/blob/master/instruments/agilent/agilent33220a.py#L1). Are these files executable, or should the shebang line be removed?

scasagrande, 2017-03-07:
You are technically correct, but I don't want to fix it.

closed 2017-03-07 by scasagrande

cgranade, 2017-03-07:
To expand a little bit, I agree if only because the shebang lines can help identify the script as Python at all to some editors which don't trust file extensions. There's no negative effect to "running" one of the internal modules, so it's likely fine to just keep them as is.