instrumentkit/InstrumentKit #161: Waiting for NDAC to go high

MichalKononenko, 2017-03-07:
I'm using the Lakeshore475 implementation provided by IK, and I get this error. What is an NDAC and why is it throwing an error.

I was able to find [this snippet](https://github.com/Galvant/gpibusb-firmware/blob/master/usb_to_gpib.c#L322) in the gpib firmware that might shed some light on it, but I would prefer a bit more context.

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/local/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "mr_freeze/__main__.py", line 43, in <module>
    loop.run()
  File "/home/corylab/.virtualenvs/MrFreeze/lib/python3.6/site-packages/mr_freeze/main_loop.py", line 89, in run
    self._run_loop(executor)
  File "/home/corylab/.virtualenvs/MrFreeze/lib/python3.6/site-packages/mr_freeze/main_loop.py", line 112, in _run_loop
    task(executor).result(self.timeout)
  File "/usr/local/lib/python3.6/concurrent/futures/_base.py", line 405, in result
    return self.__get_result()
  File "/usr/local/lib/python3.6/concurrent/futures/_base.py", line 357, in __get_result
    raise self._exception
  File "/usr/local/lib/python3.6/concurrent/futures/thread.py", line 55, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/corylab/.virtualenvs/MrFreeze/lib/python3.6/site-packages/mr_freeze/tasks/make_measurement.py", line 71, in task
    self.write_values_to_file(values_to_write, executor)
  File "/home/corylab/.virtualenvs/MrFreeze/lib/python3.6/site-packages/mr_freeze/tasks/make_measurement.py", line 77, in write_values_to_file
    self._prepare_value_for_writing(value) for value in values
  File "/home/corylab/.virtualenvs/MrFreeze/lib/python3.6/site-packages/mr_freeze/tasks/make_measurement.py", line 77, in <listcomp>
    self._prepare_value_for_writing(value) for value in values
  File "/home/corylab/.virtualenvs/MrFreeze/lib/python3.6/site-packages/mr_freeze/tasks/make_measurement.py", line 86, in _get_result
    return value.result(self.timeout)
  File "/usr/local/lib/python3.6/concurrent/futures/_base.py", line 398, in result
    return self.__get_result()
  File "/usr/local/lib/python3.6/concurrent/futures/_base.py", line 357, in __get_result
    raise self._exception
  File "/usr/local/lib/python3.6/concurrent/futures/thread.py", line 55, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/corylab/.virtualenvs/MrFreeze/lib/python3.6/site-packages/mr_freeze/tasks/report_magnetic_field.py", line 46, in task
    field_strength = self._field_strength
  File "/home/corylab/.virtualenvs/MrFreeze/lib/python3.6/site-packages/mr_freeze/tasks/report_magnetic_field.py", line 32, in _field_strength
    return self.gauge.field
  File "/home/corylab/.virtualenvs/MrFreeze/lib/python3.6/site-packages/mr_freeze/devices/lakeshore_475.py", line 77, in field
    return self._magnetometer.field
  File "/home/corylab/.virtualenvs/MrFreeze/lib/python3.6/site-packages/instruments/lakeshore/lakeshore475.py", line 99, in field
    return float(self.query('RDGFIELD?')) * self.field_units
  File "/home/corylab/.virtualenvs/MrFreeze/lib/python3.6/site-packages/instruments/lakeshore/lakeshore475.py", line 110, in field_units
    value = int(self.query('UNIT?'))
ValueError: invalid literal for int() with base 10: 'Timeout error: Waiting for NDAC to go high while writing'
```

scasagrande, 2017-03-07:
What are you using to connect to it? Are you using my adapter? NDAC is one of the handshake hardware lines for GPIB. If the bus controller is waiting for the line to go high, it means that an instrument on the bus hasn't released the line, indicating that the byte was not accepted before handshake timeout occurred.

MichalKononenko, 2017-03-07:
This is the adapter I'm using

[image]

scasagrande, 2017-03-08:
Yep that's one of mine! An older one, but still one of mine either way. It should work with the Lakshore 475; I'm pretty sure I even used that exact revision on the 475 at one point. Of course I might have broken functionality since then, but I hope not.

Make sure you've setup the line termination correctly on the instrument front panel. I believe the default is CRLF, and I don't think I supported multi-char terminators in the firmware version that board is probably running. I'd suggest starting with EOI and see what happens.

MichalKononenko, 2017-03-09:
Thanks. I'll look into it.

closed 2017-03-09 by scasagrande