instrumentkit/InstrumentKit #174: Setup issue - ValueError: ufunc <ufunc 'positive'> not supported by quantities

cmoriarty, 2017-06-19:
I created a fresh conda python2.7 enviornment, and ran pip install instrumentkit.  This worked great, although with the single import statement:

`from instruments.thorlabs import APTMotorController`

I get the following error:
```
  File "C:\ProgramData\Anaconda2\envs\kittest\lib\site-packages\quantities\quantity.py", line 236, in __array_prepare__
    """ % uf
ValueError: ufunc <ufunc 'positive'> not supported by quantities
                please file a bug report at https://github.com/python-quantities
```

scasagrande, 2017-06-22:
Hey @cmoriarty ! Thanks for the report. I missed it as I've been working on some data analysis for another project this week.

I haven't tested installation with conda actually. I'll give it a go locally, but for now can you try conda py3.4+ ? Not sure if it will help, but I'm interested to see.

cmoriarty, 2017-06-22:
No worries, I never have expectations when I post issues on github, haha.  Thanks for replying!

We are limited to python 2.7 because of a device on our system with an SDK that isn't python3 compatible (cough cough Boston Micromachines cough cough).   It sounds like you think python 2.7 is the culprit?  

Just to clarify, I did use pip (not conda) to install InstrumentKit.  I just did it in a conda environment, which is probably irrelevant.  

In the end I had to implement my own low-level interface to the d2xx USB driver to control a Thorlabs MFF101 flip motor.  I did it with the ftd2xx python wrapper on github. I will be needing to interface with a Thorlabs laser source in the next month or so, I'll check back in then.

scasagrande, 2017-06-23:
For clarification, you get the error during the installation process of `pip install instrumentkit` ? If so, it must be something related to anaconda, because IK currently installs in CI on CPython 2.7 without issue. I'll take a look this weekend.

scasagrande, 2017-06-24:
@cmoriarty It looks like this is an issue with the `quantities` package and the latest `numpy>=1.13.0`. You can see that here: https://github.com/python-quantities/python-quantities/issues/129

I'd suggest you first manually install `numpy==1.12.1` and the install InstrumentKit.

closed 2017-07-02 by scasagrande