instrumentkit/InstrumentKit #184: Interfacing Agilent 34401A

littlebrat, 2018-05-29:
I had to replace the terminator with '\r' in order to work with the same boilerplate class **Agilent34410a**. 
1. Why is this termination character different?
I am using the visa resource:
`instruments.agilent.Agilent34410a.open_visa('GPIB0::21::INSTR')`

2. Should I make a subclass that inherits **Agilent34410a** and just replace the termination character?

scasagrande, 2018-05-31:
Your instrument may have its termination character changed from default. Or you can just update it at runtime after you instantiate your instrument:

```python
inst = instruments.agilent.Agilent34410a.open_visa('GPIB0::21::INSTR')
inst.terminator = "\r"
```

scasagrande, 2018-05-31:
If it is actually a different default across all `34401A` vs `34410a`, and that is the only difference, then I suppose we should subclass it with just that terminator character being different.

closed 2018-06-05 by scasagrande