instrumentkit/InstrumentKit #185: visa_communicator is not compatible with pre-SCPI instruments.

littlebrat, 2018-06-13:
Hello,

This issue came up when I was implementing a new instrument - the HP3458A multimeter. This instrument is pre-SCPI and does not include a termination character if the response is read in bytes (as explained in https://github.com/pyvisa/pyvisa/issues/315 issue). So, the `read()` always times out when I use the VISA resource.
These are the output formats that the instrument supports: 

[image]

I need high precision, so I am using DREAL (8 bytes per reading). With this in mind I propose the following change:

### OLD (on [visa_communicator.py](https://github.com/Galvant/InstrumentKit/blob/master/instruments/abstract_instruments/comm/visa_communicator.py))

```python
            # Line 127
            while len(self._buf) < size:
                # This tries to read until a terminator is found
                data = self._conn.read()
                if data == "":
                    break
                self._buf += data
            msg = self._buf[:size]
            # Remove the front of the buffer.
            del self._buf[:size]
```

### NEW

```python
            # Line 127
            self._buf += self._conn.read_bytes(size)
            msg = self._buf[:size]
            # Remove the front of the buffer.
            del self._buf[:size]
```

This [feature](https://pyvisa.readthedocs.io/en/stable/rvalues.html#when-things-are-not-what-they-should-be) was added in the last few months on pyvisa's side.
Can I make a pull request with this change or will it break compatibility with something else?

littlebrat, 2018-06-13:
Actually, my solution does not solve the problem because the content must be unpacked using the unpack method: `struct.unpack('>d', x)`

scasagrande, 2018-06-13:
Alright, we can work with this.

So if we make the change to `GPIBCommunicator.py#L127`, we just got to put the `struct.unpack` somewhere up the chain.

If we then move up the normal chain to [`abstract_comm.py#L191`](https://github.com/Galvant/InstrumentKit/blob/master/instruments/abstract_instruments/comm/abstract_comm.py#L191), that's where we normally do our decoding. So we could do something here like this:

```python
if "binary:" in encoding:
    return struct.unpack(encoding.split(":")[1], self.read_raw(size))
else:
    return self.read_raw(size).decode(encoding)
```

Then we need `encoding` to be correctly exposed at each step up to the `Instrument` level. 

An `encoding` parameter can then be added to [`Instrument.read`](https://github.com/Galvant/InstrumentKit/blob/master/instruments/abstract_instruments/instrument.py#L154) which can default to `self._encoding`, and an `encoding` property added to the [`Instrument` class](https://github.com/Galvant/InstrumentKit/blob/master/instruments/abstract_instruments/instrument.py#L60).

A similar set of changes wil have to be done to `Instrument.query`.

littlebrat, 2018-06-13:
Ok, I did something similar to what you just told me and it works as expected. I am going to make a pull request now so that you can review the changes.

littlebrat, 2018-06-18:
@scasagrande I need to add a read termination character as shown [here](https://pyvisa.readthedocs.io/en/stable/resources.html?highlight=termination#termination-characters). Should I create new property setter/getter for that effect in [visa_communicator.py](https://github.com/Galvant/InstrumentKit/blob/master/instruments/abstract_instruments/comm/visa_communicator.py)? Right now, the termination character only adds it to the outgoing data.

scasagrande, 2018-06-18:
Are the termination characters non-symmetric? My initial reaction is that [VisaCommunicator.terminator#L87](https://github.com/Galvant/InstrumentKit/blob/master/instruments/abstract_instruments/comm/visa_communicator.py#L87) should also set the termination character down the chain to the visa resource. I'm hesitant to add a separate one just for visa because I want all the comm layers to have a consistent api.

littlebrat, 2018-06-18:
From the documentation I see that you can have different termination characters with `write_termination` and `read_termination`. The multimeter HP3458A requires me to set the read_termination character.  I'll submit my proposal to the ongoing pull request without separate setter/getter, and then you can agree/disagree with the changes 😄

timhellwig, 2018-06-29:
I have a related problem. I am currently implementing several Optical Spectrum Analyzers and trying to use binblockread to efficiently read the data trace from the device: 

```python
        # METHODS #
        
        def data(self, bin_format=True):
            cmd = ':TRAC:Y? TR'+self._name  
            self._parent.sendcmd(cmd)
            data = self._parent.binblockread(data_width = 4,fmt = '<d')
            self._parent._file.read_raw(1) #this reads a '\n'
```

with the old code in [visa_communicator](https://github.com/Galvant/InstrumentKit/blob/master/instruments/abstract_instruments/comm/visa_communicator.py)

```python
            # Line 127
            while len(self._buf) < size:
                # This tries to read until a terminator is found
                data = self._conn.read()
                if data == "":
                    break
                self._buf += data
            msg = self._buf[:size]
            # Remove the front of the buffer.
            del self._buf[:size]
```

I get 
```
    a = osa.channel[0].data()
  File "m:\software\instrumentkit\instruments\yokogawa\yokogawa6370.py", line 83, in data
    data = self._parent.binblockread(data_width = 4,fmt = '<d')
  File "m:\software\instrumentkit\instruments\abstract_instruments\instrument.py", line 277, in binblockread
    symbol = self._file.read_raw(1)
  File "m:\software\instrumentkit\instruments\abstract_instruments\comm\visa_communicator.py", line 128, in read_raw
    data = self._conn.read()
  File "C:\ProgramData\Anaconda3\lib\site-packages\pyvisa\resources\messagebased.py", line 413, in read
    message = self._read_raw().decode(enco)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xf6 in position 7: ordinal not in range(128)
```
This happens when binblockread tries to read the # character to determine if it really is a binary block following. Using read_bytes() instead of read() in the visa_communicator also fixes this issue.

scasagrande, 2018-06-29:
Can you try the changes in his branch and see if that solves your issue?

timhellwig, 2018-06-30:
I merged this branch into my [osa branch](https://github.com/timhellwig/InstrumentKit/tree/optical_spectrum_analyzers) and it solves my issue without new errors or problems.

scasagrande, 2019-02-07:
Fix merged in PR #192

closed 2019-02-07 by scasagrande