instrumentkit/InstrumentKit #188: Cannot connect to APTPiezoStage with open_serial

dupuisni, 2018-09-19:
Hi,

I am having some issues to use `APTPiezoStage`. I am trying to connect using `open_serial` which always raise `TypeError: can't concat str to bytes`. I tried connecting to other non-APT Thorlabs instruments using the same method and that works great. The instrument I am trying to connect is a 3-channel closed-loop Piezo controller (BPC303).

scasagrande, 2018-09-19:
Can you give me a full traceback so I can see where the TypeError is being raised?

crazy4pi314, 2018-09-19:
Which version of python are you using? I have had problems with string encoding with the changes in 3, and have a fix I need to PR, but haven't bc I didn't get it fully tested yet.

dupuisni, 2018-09-19:
I use python 3.7. Also tried with 3.4, and 3.6 but no luck. Here is the full traceback:

Traceback (most recent call last):
  File "ex_BPC303.py", line 9, in <module>
    boa = BPC303.open_serial(vid=vid, pid=pid, baud=115200)
  File "/home/ndupuis/Documents/testing_Thorlabs/py37_ik/lib/python3.7/site-packages/instrumentkit-0.1.1-py3.7.egg/instruments/abstract_instruments/instrument.py", line 525, in open_serial
  File "/home/ndupuis/Documents/testing_Thorlabs/py37_ik/lib/python3.7/site-packages/instrumentkit-0.1.1-py3.7.egg/instruments/thorlabs/bpc303.py", line 34, in __init__
  File "/home/ndupuis/Documents/testing_Thorlabs/py37_ik/lib/python3.7/site-packages/instrumentkit-0.1.1-py3.7.egg/instruments/thorlabs/thorlabsapt.py", line 110, in __init__
  File "/home/ndupuis/Documents/testing_Thorlabs/py37_ik/lib/python3.7/site-packages/instrumentkit-0.1.1-py3.7.egg/instruments/thorlabs/_abstract.py", line 59, in querypacket
  File "/home/ndupuis/Documents/testing_Thorlabs/py37_ik/lib/python3.7/site-packages/instrumentkit-0.1.1-py3.7.egg/instruments/abstract_instruments/instrument.py", line 134, in query
  File "/home/ndupuis/Documents/testing_Thorlabs/py37_ik/lib/python3.7/site-packages/instrumentkit-0.1.1-py3.7.egg/instruments/abstract_instruments/comm/abstract_comm.py", line 233, in query
  File "/home/ndupuis/Documents/testing_Thorlabs/py37_ik/lib/python3.7/site-packages/instrumentkit-0.1.1-py3.7.egg/instruments/abstract_instruments/comm/serial_communicator.py", line 191, in _query
  File "/home/ndupuis/Documents/testing_Thorlabs/py37_ik/lib/python3.7/site-packages/instrumentkit-0.1.1-py3.7.egg/instruments/abstract_instruments/comm/abstract_comm.py", line 218, in sendcmd
  File "/home/ndupuis/Documents/testing_Thorlabs/py37_ik/lib/python3.7/site-packages/instrumentkit-0.1.1-py3.7.egg/instruments/abstract_instruments/comm/serial_communicator.py", line 175, in _sendcmd
TypeError: can't concat str to bytes

scasagrande, 2018-09-19:
`instrumentkit-0.1.1`

are you sure you have the latest version installed

dupuisni, 2018-09-19:
Indeed, for that test I branched from ch-thorlabs-mdt693B, did not realized that it was not the latest version. Here is the full traceback with the latest version:

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
/home/ndupuis/Documents/testing_Thorlabs/InstrumentKit/test_ik.py in <module>()
      1 import instruments as ik
      2
----> 3 test=ik.thorlabs.APTPiezoStage.open_serial(vid=int('0403',16),pid=int('faf0',16))

/home/ndupuis/Documents/testing_Thorlabs/InstrumentKit/instruments/abstract_instruments/instrument.py in open_serial(cls, port, baud, vid, pid, serial_number, timeout, write_timeout)
    527             write_timeout=write_timeout
    528         )
--> 529         return cls(ser)
    530
    531     @classmethod

/home/ndupuis/Documents/testing_Thorlabs/InstrumentKit/instruments/thorlabs/thorlabsapt.py in __init__(self, filelike)
    108             )
    109             hw_info = self.querypacket(
--> 110                 req_packet, expect=_cmds.ThorLabsCommands.HW_GET_INFO)
    111
    112             self._serial_number = str(hw_info.data[0:4]).encode('hex')

/home/ndupuis/Documents/testing_Thorlabs/InstrumentKit/instruments/thorlabs/_abstract.py in querypacket(self, packet, expect)
     57         :rtype: `ThorLabsPacket`
     58         """
---> 59         resp = self.query(packet.pack())
     60         if not resp:
     61             if expect is None:

/home/ndupuis/Documents/testing_Thorlabs/InstrumentKit/instruments/abstract_instruments/instrument.py in query(self, cmd, size)
    132
    133         if ack_expected_list[0] is None:  # Case no ACK
--> 134             value = self._file.query(cmd, size)
    135         else:  # Case with ACKs
    136             _ = self._file.query(cmd, size=0)  # Send the cmd, don't read

/home/ndupuis/Documents/testing_Thorlabs/InstrumentKit/instruments/abstract_instruments/comm/abstract_comm.py in query(self, msg, size)
    231         if self.debug:
    232             self._logger.debug(" <- %s", repr(msg))
--> 233         resp = self._query(msg, size)
    234         if self.debug:
    235             self._logger.debug(" -> %s", repr(resp))

/home/ndupuis/Documents/testing_Thorlabs/InstrumentKit/instruments/abstract_instruments/comm/serial_communicator.py in _query(self, msg, size)
    189         :rtype: `str`
    190         """
--> 191         self.sendcmd(msg)
    192         return self.read(size)

/home/ndupuis/Documents/testing_Thorlabs/InstrumentKit/instruments/abstract_instruments/comm/abstract_comm.py in sendcmd(self, msg)
    216         if self.debug:
    217             self._logger.debug(" <- %s", repr(msg))
--> 218         self._sendcmd(msg)
    219
    220     def query(self, msg, size=-1):

/home/ndupuis/Documents/testing_Thorlabs/InstrumentKit/instruments/abstract_instruments/comm/serial_communicator.py in _sendcmd(self, msg)
    173         :param str msg: The command message to send to the instrument
    174         """
--> 175         msg += self._terminator
    176         self.write(msg)
    177

TypeError: can't concat str to bytes

scasagrande, 2018-09-19:
Looks like we might have to change `ThorLabsPacket` and change `resp = self.query(packet.pack())` to use `send_raw` as the underlying implementation of `query` uses `sendcmd`, which is looking for a `str` input. I believe `packet.pack()` is `bytes`, and I imagine we shouldn't be appending a termination character anyways.

crazy4pi314, 2018-09-19:
Let me look at my stuff tonight, I did fix this at one point for talking to the k100 cubes. Ill make a new PR then to fix it hopefully then.

dupuisni, 2018-09-19:
Looks like the issue has been solved already in `feature-apt-tables` branch. I just tested and no more TypeError.

scasagrande, 2018-09-20:
I'll try to cherry pick that fix into master then 👌

scasagrande, 2019-02-07:
I've resurrected PR #167 in an attempt to address this issue. The biggest functional difference is going to be master branch has been merged in. If I don't end up hearing anything I'll merge it into `develop` shortly to be staged with a few other changes for a `0.5.0` release.

scasagrande, 2019-02-10:
I haven't heard anything, so that branch is now merged into the develop branch.

closed 2019-02-10 by scasagrande

sashagaz, 2019-08-06:
@dupuisni & @scasagrande can you please provide an example or explanation how to instantiate serial interface to instruments.thorlabs.APTPiezoStage. I don't see a serial method in the piezo class.

scasagrande, 2019-08-07:
All instruments inherit from the base `Instrument` class.

The steps to instantiate any of the instrument drivers to any of the supported connection methods is the same.

sashagaz, 2019-08-07:
Thanks for the clarification. I tried this:

inst = thor.APTMotorController.open_serial("com16",115200)
x = inst.MotorChannel(inst,1)
x.go_home()

but I am getting a timeout error.

Any help would be much appreciated.