instrumentkit/InstrumentKit #190: Todo List for v1

scasagrande, 2019-02-06:
There's a few breaking changes that I would like to get in all at once in the move to v1. This is on my priority list for this year. The list is subject to change, but will likely not get smaller.

- [ ] Drop Py27 support.
- [ ] Convert to using `pint` instead of `quantities` #92 
- [ ] Remove numpy as a requirement #91

bilderbuchi, 2019-02-07:
I think aiming for one probably "architectural" issues should also be included, e.g. #15 (which already is in the milestone) and #156.
p.s. great to see activity again! :-)

scasagrande, 2019-02-07:
I wasn't able to add more instrument classes last year due to the nature of my day job. If I added ones for equipment I use, I could have revealed extra info on what we're working on 😱 . But we're past the announce hurdle and so I'm back at it here 😎 .

(If you're curious, check it out at bynorth.com)

bilderbuchi, 2020-02-25:
> (If you're curious, check it out at bynorth.com)

I think Willam Gibson's latest book [Agency](https://www.goodreads.com/book/show/34943643-agency) will be interesting for you. ;-)

scasagrande, 2020-05-06:
@bilderbuchi heh, I ended up leaving that company shortly after that post. I now lead the test automation effort at [SSIMWAVE](https://www.ssimwave.com/)

scasagrande, 2020-12-19:
Maybe time for us to start thinking about anything else we might have to go here? While the remaining items in the list would be nice to have, I don't want to keep pushing out a new release just to get more things in that can easily wait.

- [ ] Migrate to new org #244
- [ ] Migrate to github actions (thanks travis!)

trappitsch, 2020-12-19:
Makes sense to do a release, there have been quite some really neat updates since v0.6.0. Async would be cool for v1. Do you want to jump straight to v1 or do a v0.7.0 with the updates? 

Inner vs. outer class consistency shouldn't be too difficult, I could help out with that one. Seems like you have started on multi-channel consistency a while ago, how far is that, or was that actually completed?

scasagrande, 2020-12-20:
I don't think 0.7.0 would be a good idea. Switching to `pint` is a breaking change and is the main reason to be going to `1.x`. Async features would build on the existing and add new optional functionality, and so shouldn't need a major version bump.

Of course, we could also have a conversation about going to calver.

trappitsch, 2020-12-20:
Oh, yeah, breaking change slipped my mind. Also: dropped py27, so `1.x` is definitely the way to go...

> Of course, we could also have a conversation about going to calver.

Could :slightly_smiling_face:

scasagrande, 2022-01-27:
I'm going to ship a pre-release version `1.0.0.dev0`

Then lets tackle the work to move to its own organization, and then start reviewing docs, examples, etc.

scasagrande, 2022-01-27:
`1.0.0.dev0` now up on pypi :tada:  you can use the `--pre` flag for `pip install` to install it