instrumentkit/InstrumentKit #199: Drop support for Py27

scasagrande, 2019-02-07:
The time is coming near! Py27 is finally reaching EOL in ~1 year. I want to have Py27 support removed for the v1.0 version.

This issue will contain a list of things that will need to be done to go with this. This list isn't finalized, and might expand as time goes.

- [ ] Remove `from __future__ import X` statements
- [ ] Remove inheritance from `object`
- [ ] Remove `enum34` package
- [ ] Change `setup.py` to indicate only Py3 is supported
- [ ] Update documentation to mention only Py3 is supported
- [ ] Change class definitions that utilize metaclasses to use the Py3 specific syntax (`metaclass=` instead of using `future.utils.with_metaclass`)

jaron-l, 2020-01-02:
I know I'm early to the game but Python support for 2.7 stopped yesterday. Currently enum34 is breaking pyinstaller so when enum34 is removed, instrumentkit can work with pyinstaller theoretically.

scasagrande, 2020-01-02:
Oh interesting

And don't worry, you're not early lol. I was cracking the whip today on everyone at work, so now its my turn for myself.

scasagrande, 2020-01-13:
@jaron-l I've removed enum34 on master so feel free to give that a try

closed 2020-02-04 by scasagrande