instrumentkit/InstrumentKit #212: Adding Additional RF Instrument Support

JAnderson419, 2019-02-21:
I have some code I wrote for our group's test equipment that is fairly crude in its implementation (little input checking, relies on user to know specs of instrument, no general classes - see https://github.com/hybridMEMS/PyMeasRF). The code includes instrument classes to implement basic SCPI commands through pyvisa, as well as a couple different functions to execute measurements our group makes that integrate several instruments (DC biasing of RF devices for s-parameter measurements, quick and dirty pseudo-parameter analyzer testing of semiconductor devices by tying together discrete SMUs, etc). Rather than continuing to add to it on my own (implementing general instrument classes and more robust error handling/unit tests), I would like to try to integrate with IK to have python instrument control more centralized.

I may be able to port over after reviewing your dev guide some more, however, it seems that there are not yet generic instrument classes for some of the equipment (SMU, spectrum analyzer, network analyzer). I suppose a generic SMU may be able to inherit multimeter and power_supply? Additionally, I am not sure of the best place for the "experiments" section of my code. I see some templates for basic measurements on individual instrument pages in the docs, but I don't see examples of integration of instruments into measurement setups. I know these may not be as widely useful as specific instrument examples, but I am happy to include them in case they are of use to others. 

- [ ] Create Generic Classes 
- [ ] Implement instrument settings as properties
- [ ] Port over methods, using the properties scheme as an intermediary between the user-friendly method and SCPI calls. 
- [ ] Update docstrings from numpy style to PEP.
- [ ] Create unit tests

On an unrelated note, I have access to various equipment at work and home that I may be able to test generic instruments against if that is of any use.

-Jackson

scasagrande, 2019-02-22:
Hi Jackson! I'm happy to hear you'd like to contribute. I'll take a look at your code and see what you have done, then maybe I'll start getting some generic base classes implemented so we can come up with an API.

Re:"experiments", from first glance I don't think that's something we would put into IK. We do have an examples folder where some of that might fit in, but your actual experiment software is going to be very specific to your situation. In the end, we'd have something like the instrument drivers in IK, a small example included in the repo, and then your specific application in your own repository.

JAnderson419, 2019-02-23:
That's a smart approach  to keep the examples separate - keeps the size of the code down for people who would probably never use most of it. That being said, while not directly related to instrument drivers, I think having a library of implementation examples is a good driver of adoption. The hard part for any of these libraries is to lower the barrier of entry enough for test engineers that reading python package documentation and implementing/writing drivers for them is less complicated than sitting down with a programming manual and skimming through pages of commands. Such a discussion is probably more appropriate for @bilderbuchi's LabPy group, but I don't know that he has many people following it right now.

I'm also going to page @mvddf as he is another user who has shown interest in using Python to control RF test equipment in the past. I believe the two of us have a good mix of Keysight RF stuff (PNA, PXA, PSG, MXA, SSA) and could perhaps help in writing tests and/or checking compatibility of generic classes for these instruments? All of the high end Keysight stuff is a mess of software options that significantly alter the capabilities of the tools, but at the very least we should be able to enable basic functionality as a start.

scasagrande, 2019-02-24:
Oh we do have an examples folder in the current repo: https://github.com/Galvant/InstrumentKit/tree/master/doc/examples

bilderbuchi, 2019-04-05:
> Such a discussion is probably more appropriate for @bilderbuchi's LabPy group, but I don't know that he has many people following it right now.

yeah, no, unfortunately I could never entice enough people to contribute to obtain a critical mass to get this space moving in the right direction. :-/ you are very welcome to contribute, though, at least the information gets preserved for posterity!

vehstedt, 2019-11-05:
Have @JAnderson419 or @bilderbuchi looked at QCoDeS? It's mostly geared towards nanoelectronics measurements, but there are controls for some RF instruments (Keysight/HP, R&S): 
https://github.com/QCoDeS/Qcodes/tree/master/qcodes/instrument_drivers

vehstedt, 2019-11-05:
Unfortunately, qcodes does not currently support Thorlabs motorised stages (specially not on linux), which is how I ended up here. Suggestions/references for adding platform-independent APT controls would be welcome!