instrumentkit/InstrumentKit #217: Problem Communicating with Thorlabs LCC25

byurash, 2019-06-25:
Something is wrong with the LCC25 instrument class. I suspect it has something to do with the termination character.  The 'name' command cycles through several outputs. The first time it outputs 'THORLABS LCC25 VERS 1.01' which is promising. Repeat the 'name' command and now I get a 'AcknowledgementError: Incorrect ACK message received: got  *idn? expected *idn?'
Repeat the 'name' command a 3rd time and the error slightly changes: 'AcknowledgementError: Incorrect ACK message received: got THORLABS LCC25 vers 1.01 expected *idn?'
Repeat the 'name' command a 4th time and I get: 'AcknowledgementError: Incorrect ACK message received: got > *idn? expected *idn?'
5th time and we are back to: 'Incorrect ACK message received: got THORLABS LCC25 vers 1.01 expected *idn?'
The '>' character is in the Thorlabs manual, they say this is the command prompt. So it seems that maybe InstrumentKit doesn't recognize that as the beginning of each read() command.
Something is just a bit off... the connection is definitely established, I can enable the instrument from time to time, but it always has an 'AcknowledgementError'

scasagrande, 2019-07-24:
Hi there, sorry for the delay. New day job and all.

The Ack error is because the LCC25, during our original development, would echo the sent command back to the computer, and so we implemented a way to capture and verify that ACK message. Perhaps your instrument is not set to send that echo ACK back to the computer.

The `>` prompt character is handled differently and is raised as a separate error if it is not present.

trappitsch, 2022-01-23:
This looks very much related and almost a duplicate of #310 which should have been fixed in #311.

scasagrande, 2022-01-24:
Good call

closed 2022-01-24 by scasagrande