instrumentkit/InstrumentKit #218: example code for python Thorlabs LCC25 doesn't work

byurash, 2019-06-25:
The example script for Thorlabs LCC25 doesn't work - I'd love to see someone's code that actually controlled the instrument reliably!

bilderbuchi, 2019-06-26:
I think a more precise description could be valuable to whoever will look into this - what does "doesn't work" mean? What breaks or does not work as expected? Do you get error messages?

byurash, 2019-06-26:
Sorry, was too frustrated at the moment of writing to explain.
First off, I believe line 7 has a typo. In the right-hand-side of the equality 'llc' should be changed to 'lcc'.
The script I ran is shown below (just 3 lines):

import instruments as ik
lcc = ik.thorlabs.LCC25.open_serial('COM6', 115200,timeout=1)
lcc.mode = lcc.Mode.voltage1

Running the script several times results in the following AcknowledgementErrors:
1) no error

2)   File "C:\Users\bayurash\AppData\Local\Continuum\anaconda3\lib\site-packages\instruments\abstract_instruments\instrument.py", line 107, in sendcmd
    "expected {}".format(ack, ack_expected)
AcknowledgementError: Incorrect ACK message received: got  mode=1 expected mode=1

3)   File "C:\Users\bayurash\AppData\Local\Continuum\anaconda3\lib\site-packages\instruments\abstract_instruments\instrument.py", line 107, in sendcmd
    "expected {}".format(ack, ack_expected)
AcknowledgementError: Incorrect ACK message received: got > mode=1 expected mode=1

4)   File "C:\Users\bayurash\AppData\Local\Continuum\anaconda3\lib\site-packages\instruments\abstract_instruments\instrument.py", line 107, in sendcmd
    "expected {}".format(ack, ack_expected)
AcknowledgementError: Incorrect ACK message received: got > mode=1 expected mode=1

scasagrande, 2019-07-24:
It looks like there might be something stuck in the buffer, or the instrument is in a slightly different mode vs that of the original author. 

My guess is that in the first error message is there there is a different in whitespace that is hard to see in the console.

Check that your instrument is set to the matching termination character. I believe we set the default for the LCC25 to `\r` (CR)