instrumentkit/InstrumentKit #241: 'decibel-milliwatt has already been registered for 1 dBm (decibel-milliwatt)'

dso2ng, 2020-07-06:
Running from within **`jupyter/scipy-notebook:latest`** docker image, the host is `Ubuntu 20.04 LTS`, the docker environment is run by issuing:
**`docker  run -p 8888:8888 -e JUPYTER_ENABLE_LAB=yes -v `pwd`:`pwd` -w `pwd` -it --name jupyterlab jupyter/scipy-notebook:latest`**
The installation of this package was done interactively by:
**`docker exec -it jupyterlab pip install git+git://github.com/Galvant/InstrumentKit@master`**

After that from within the Jupyterlab notebook the module was imported:
**`from InstrumentKit import instruments as ik`**

Then I got error complaining about **`decibel-milliwatt registration`**, the traceback message listed below:

```python traceback
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
/opt/conda/lib/python3.7/site-packages/quantities/registry.py in __setitem__(self, string, val)
     30             try:
---> 31                 assert string not in self.__context
     32             except AssertionError:

AssertionError: 

During handling of the above exception, another exception occurred:

KeyError                                  Traceback (most recent call last)
<ipython-input-3-98114edbb150> in <module>
----> 1 from InstrumentKit import instruments as ik

/home/dso2ng/OneDrive/jupyter/InstrumentKit/instruments/__init__.py in <module>
     32 from . import yokogawa
     33 
---> 34 from . import units
     35 from .config import load_instruments
     36 

/home/dso2ng/OneDrive/jupyter/InstrumentKit/instruments/units.py in <module>
     30 
     31 #: Decibel-milliwatts, a basic unit of logarithmic power.
---> 32 dBm = UnitLogPower('decibel-milliwatt', symbol='dBm')
     33 
     34 # The Phase Matrix signal generators communicate in units of millihertz (mHz)

/opt/conda/lib/python3.7/site-packages/quantities/unitquantity.py in __init__(self, name, definition, symbol, u_symbol, aliases, doc)
    313     ):
    314         super(IrreducibleUnit, self).__init__(
--> 315             name, definition, symbol, u_symbol, aliases, doc
    316         )
    317         cls = type(self)

/opt/conda/lib/python3.7/site-packages/quantities/unitquantity.py in __init__(self, name, definition, symbol, u_symbol, aliases, doc)
     76         aliases=[], doc=None
     77     ):
---> 78         unit_registry[name] = self
     79         if symbol:
     80             unit_registry[symbol] = self

/opt/conda/lib/python3.7/site-packages/quantities/registry.py in __setitem__(self, label, value)
     46 
     47     def __setitem__(self, label, value):
---> 48         self.__registry.__setitem__(label, value)
     49 
     50     def __getitem__(self, label):

/opt/conda/lib/python3.7/site-packages/quantities/registry.py in __setitem__(self, string, val)
     35                 raise KeyError(
     36                     '%s has already been registered for %s'
---> 37                     % (string, self.__context[string])
     38                 )
     39             self.__context[string] = val

KeyError: 'decibel-milliwatt has already been registered for 1 dBm (decibel-milliwatt)'
```

trappitsch, 2020-07-07:
Why to you run: `from InstrumentKit import instruments as ik`? You do the `pip install` of the package, so you should be able to import the package as:

    import instruments as ik

I tried to reproduce your setup outside of docker, as far as I can follow it at least and can import as just described. However, if I clone the repo and then call `from InstrumentKit import instruments as ik`, I get the same error. InstrumentKit is not the package, instruments is. So I would in fact expect that this import would run into issues.

scasagrande, 2020-07-07:
@trappitsch has the correct answer, change your import statement and you should be good.

closed 2020-07-07 by scasagrande