instrumentkit/InstrumentKit #244: Migrate instrumentkit to its own organization

scasagrande, 2020-07-21:
I think its time to split the project off from its current org into its own. It'll make project organization a little easier, as well as when I/we eventually split the instrument drivers out into their own repository.

scasagrande, 2020-07-21:
I have created the instrumentkit org https://github.com/instrumentkit but going to hold off in case there is comments / concerns.

trappitsch, 2020-07-22:
I think that's a great idea. I have in fact recently started the same for some projects. Putting it in its own organization allows for more granular rights, especially now that teams and their management are free (I think at least that's new).

Might be valuable to keep a check list on what needs to be updated / tested after a move of the whole project. I don't have much experience for moving large projects around, everything I moved so far had only a handful of people using it.

> as well as when I/we eventually split the instrument drivers out into their own repository.

I guess for this task it would definitely be very valuable.

scasagrande, 2020-07-22:
Yeah. My current thought process is to use something like pluggy to allow for people and orgs to maintain their own set of instrument drivers, while being able to make use of the IK core + collection of tested included drivers.

trappitsch, 2020-07-22:
That idea sounds great. I guess this would also improve maintainability?
I guess your first / current step (especially with the recent WIP PR #243) is still to prep for a v1 release #190?

scasagrande, 2020-07-22:
Yes

Although moving the project to its own org is independent of that. But will be done sooner rather than later.

scasagrande, 2020-09-01:
First thing I noticed, I will have to give travis and coveralls.io access to the new org

bilderbuchi, 2020-11-16:
Watch out with Travis, though, the [new terms of service](https://blog.travis-ci.com/2020-11-02-travis-ci-new-billing) mean, as far as I can tell, that you have to re-request build minutes when they run out.

bilderbuchi, 2020-11-16:
we (that is: @cjermain) just did this for pymeasure ([new home](https://github.com/pymeasure/pymeasure)), it went smoothly; you can put "update readthedocs builders" and "changing your own git remote config" on the list.

trappitsch, 2020-12-05:
> use something like pluggy

Came across this article and thought I'll share it since pluggy was mentioned in this thread here.
https://kracekumar.com/post/build_plugins_with_pluggy/

bilderbuchi, 2020-12-09:
Watch out, the Travis OSS situation seems to be deteriorating: https://news.ycombinator.com/item?id=25338983

scasagrande, 2020-12-09:
Yeahhhh, I noticed that as well. Sounds like we might be migrating yet again.

trappitsch, 2020-12-09:
Could switch it over to Github Actions. I've been using them on another OSS project and integration is awesome, otherwise work great too. Shouldn't be too complicated...

scasagrande, 2020-12-09:
Yeah that's what I'm eyeing right now. I took a look at my travis credits
for IK and we've used about 25%. The files look similar so it shouldn't be
too difficult.

Heck, it probably integrates nicely with dependabot!

bilderbuchi, 2020-12-09:
Feel free to look at the CI config for [pymeasure](https://github.com/pymeasure/pymeasure/tree/master/.github). I can try to answer questions if you have any. There are a couple open things/bugs in the tooling, but it's already an improvement compared to Travis. 
I like the possibility to decorate changes, for example one can use linter output to decorate the lines of a PR that triggered it - so far I only did that for sphinx, though.

scasagrande, 2022-03-29:
Moved to the new `instrumentkit` org! I'm not sure if coveralls is struggling right now, or if I have something misconfigured.

Either way, its moved :joy: :tada:

closed 2022-03-29 by scasagrande