instrumentkit/InstrumentKit #273: pyvisa deprecation warning during test run

scasagrande, 2020-10-14:
Warning message is as follows:

FutureWarning: The visa module provided by PyVISA is being deprecated. You can replace `import visa` by `import pyvisa as visa` to achieve the same effect.

trappitsch, 2020-10-15:
Saw that warning. Looks like it's just a potential naming conflict with [https://github.com/visa-sdk/visa-python](https://github.com/visa-sdk/visa-python). The `import pyvisa as visa` fix is thus probably okay, what do you think?

bilderbuchi, 2020-10-15:
Indeed that `visa` package is the reason, we also hit this already and fixed it like [this](https://github.com/ralph-group/pymeasure/commit/c6c6ffa952425dbd75cbf41e7234210d6cc9076d).

scasagrande, 2020-10-15:
Yeah I would just fully do the rename such that it is `import pyvisa` as its not like we have a ton of occurances of it.

trappitsch, 2020-10-15:
For sure cleaner that way. I got some time to do that if you like, should be simple enough.

scasagrande, 2020-10-15:
Sure go ahead. I'm updating the pint branch right now.

scasagrande, 2020-10-15:
resolved

closed 2020-10-15 by scasagrande