instrumentkit/InstrumentKit #298: Pyserial 3.5 breaks the build

scasagrande, 2020-12-08:
`pyserial==3.5` was just released, and it seems to have broken the build with the following test failures:

```======================================================================= short test summary info ========================================================================
FAILED instruments/tests/test_base_instrument.py::test_instrument_open_serial_by_usb_ids_multiple_matches - TypeError: __init__() missing 1 required positional argum...
FAILED instruments/tests/test_base_instrument.py::test_instrument_open_serial_by_usb_ids - TypeError: __init__() missing 1 required positional argument: 'device'
FAILED instruments/tests/test_base_instrument.py::test_instrument_open_serial_by_usb_ids_incorrect_serial_num - TypeError: __init__() missing 1 required positional a...
FAILED instruments/tests/test_base_instrument.py::test_instrument_open_serial_by_usb_ids_and_serial_number - TypeError: __init__() missing 1 required positional argu...
FAILED instruments/tests/test_base_instrument.py::test_instrument_open_serial_by_usb_ids_cant_find - TypeError: __init__() missing 1 required positional argument: 'd...
```

trappitsch, 2020-12-08:
Looked a bit more into this. The issue is the mock class that is introduced in `test_base_instrument`. 

In pyserial==3.5 -> serial -> tools -> list_ports_common.py, the `ListPortsCommon` class has the initalilzer:

```python
def __init__(self, device, skip_link_detection=False):
```

In pyserial==3.4, the device was initialize as `device=None`.

Mocking serial ports in the tests, `ik` so far initialized the `ListPortsCommon` w/o a device and then assigned it later with a property call. This does not work anymore, the device must now be initialized with `ListPortsCommon`. Just testing the fix, seems to be working.

scasagrande, 2020-12-09:
Resolved! :tada:

closed 2020-12-09 by scasagrande