instrumentkit/InstrumentKit #310: Problem with LCC25 communication

MarChan13, 2021-09-17:
I am trying to communicate simple commands to the LCC25 but different errors occur. Here there is the code I run:

import instruments
from instruments.thorlabs import LCC25

lcc=LCC25.open_serial('COM3', 115200,timeout=1)
lcc.voltage1=2;
print(lcc.name)

The first time I run the code I get this error and voltage1 is set at 2 V: 

1)  File "....\lib\site-packages\instruments\abstract_instruments\instrument.py", line 140, in query raise AcknowledgementError(
AcknowledgementError: Incorrect ACK message received: got  *idn? expected *idn?

then I get:
2)  File "...anaconda3\lib\site-packages\instruments\abstract_instruments\instrument.py", line 105, in sendcmd raise AcknowledgementError( AcknowledgementError: Incorrect ACK message received: got THORLABS LCC25 vers 1.01 expected volt1=2.0

3)  File "C:\Users\marwa\anaconda3\lib\site-packages\instruments\abstract_instruments\instrument.py", line 105, in sendcmd
    raise AcknowledgementError( AcknowledgementError: Incorrect ACK message received: got > volt1=2.0 expected volt1=2.0

4)  File "C:\Users\marwa\anaconda3\lib\site-packages\instruments\abstract_instruments\instrument.py", line 105, in sendcmd
    raise AcknowledgementError( AcknowledgementError: Incorrect ACK message received: got > volt1=2.0 expected volt1=2.0

and so on...
I guess there is a problem with the reading\writing of the commands...
How can I fix it?

trappitsch, 2021-09-22:
Hi there! Sorry for the late reply, I was traveling and just got back. Looks like there's an issue with the handling of the acknowledgement and prompt. I was briefly wondering if it might be similar to #296, but I don't fully understand all the error messages you get. Could you run two examples and post the output:
Example 1:
```python
import instruments
from instruments.thorlabs import LCC25
lcc=LCC25.open_serial('COM3', 115200,timeout=1)
lcc.voltage1=2
```

Example 2:
```python
import instruments
from instruments.thorlabs import LCC25
lcc=LCC25.open_serial('COM3', 115200,timeout=1)
print(lcc.voltage1)
```

I know, these are very similar to the examples that you gave above, but I am not fully sure what error goes with what command.
The LCC25 class is a bit older, I'm sure @scasagrande has some input here as well.

MarChan13, 2021-09-24:
I tried with the first example and this is what i got:

1) run 1

voltage 1 turned to 2V with this message: The visa module provided by 
PyVISA is being deprecated. You can replace `import visa` by `import 
pyvisa as visa` to achieve the same effect.
The reason for the deprecation is the possible conflict with the visa 
package provided by the https://github.com/visa-sdk/visa-python which 
can result in hard to debug situations.
   warnings.warn(

2) run 2

error message:
  line 105, in sendcmd
     raise AcknowledgementError(AcknowledgementError: Incorrect ACK 
message received: got  volt1=2.0 expected volt1=2.0

Example 2:

line 141, in query
     raise AcknowledgementError(AcknowledgementError: Incorrect ACK 
message received: got volt1? expected volt1?

Thank you again for your time and help

Best regards,

MC

trappitsch, 2021-09-24:
Interesting, this helps. First of all, it looks like you are using the release version that is currently on pip. Could you install the current version that is on github? That means, uninstall instrumentkit and install it again using:

```
pip install git+https://github.com/Galvant/InstrumentKit.git
```

This will give you the latest version from GitHub, there have been quite some changes and more are coming, therefore there hasn't been a new release yet. That will at least take care of the warning regarding `pyvisa`, I don't think it will necessarily take care of the actual error messages, but could you please double check?

If the problems don't get fixed, please try the following: After opening the instrument in the above examples and before setting or querying anything in the above examples, insert the following line:

```python
lcc.prompt = "> "
```
Important here is the white space after `>` before closing with `"`.
It looks from the error messages that the missing character is a whitespace. We saw that before for `sc10.py`, see #296. 

Let us know how it goes!

MarChan13, 2021-09-24:
It works now, by adding the white space.

Thank you very much!!

trappitsch, 2021-09-24:
Awesome, thanks for confirming. This means it's really the exact same issue as #296. Since @MarChan13 has the hardware, we should probably go ahead as we did back then and change the prompt to `"> "` with a white space after the `>` and before the `"`. @scasagrande: I can prep a PR for this and adjust the tests to agree with the issue.