instrumentkit/InstrumentKit #331: Can't set the temperature on TC200 Thorlabs

njrebelo, 2022-02-18:
[image]

I have written this code:

import instruments as ik

com='COM7'
baud_rate = 115200

temperature_sensor=ik.thorlabs.TC200.open_serial(com,baud_rate)
print(temperature_sensor.temperature)

temperature_sensor.temperature_set(35)

and no matter what I do the erros is always the same:
Traceback (most recent call last):
  File "C:\Users\mc2\Desktop\TC200_automation.py", line 9, in <module>
    temperature_sensor.temperature_set(35)
  File "C:\Users\mc2\AppData\Local\Programs\Python\Python36\lib\site-packages\instruments\thorlabs\tc200.py", line 198, in temperature_set
    return float(response) * pq.degC
ValueError: could not convert string to float: ' 30.0elsius'

What can I do?

scasagrande, 2022-02-18:
`temperature_set` is a property, not a function to be called

try `temperature_sensor.temperature_set = 35`

trappitsch, 2022-02-18:
Looks to me like your device is sending back the string `30 Celsius` when you run `print(temperature_sensor.temperature)`.
With your code initialized as above in the image, up to line 6, could you run the following lines individually and post the output (error or not):

```python
print(temperature_sensor.query("tset?"))
print(temperature_sensor.query("tact?"))
```

Also: what version of `instrumentkit` do you have installed?

scasagrande, 2022-02-18:
Or maybe its just that `tact` returns with the short form of the units, while `tset?` returns the long form?

scasagrande, 2022-02-18:
Also this is likely 0.6 based on `pq.degC` instead of `u.degC`

trappitsch, 2022-02-18:
@scasagrande Good point, I missed that return in the output. Also about the version. So maybe @njrebelo could still run the two queries I posted and try to set the temperature as @scasagrande pointed out above. 

If it really sends back mixed messages (`" C"` for some and `" Celsius"`), it wouldn't be too hard a fix to implement.

scasagrande, 2022-02-18:
@trappitsch agreed

if I get a moment this weekend I'll take a look in the manual and see what it says

scasagrande, 2022-02-19:
So in typical Thorlabs fashion the manual is very unhelpful. It doesn't give example return values for the queries.

trappitsch, 2022-02-19:
That sounds about right... Hopefully @njrebelo can send some typical output for `tset and `tact`

njrebelo, 2022-02-21:
Thank you so much for all the help, you guys were super helpful.
I its running now as I want it to.

[image]

scasagrande, 2022-02-21:
@njrebelo could you still try this for us? `print(temperature_sensor.query("tset?"))`

njrebelo, 2022-02-28:
[image]
Sorry wasn´t in the lab much this week

scasagrande, 2022-02-28:
np @njrebelo , thank you for getting us that data!

trappitsch, 2022-03-01:
Hmm, this is interesting and weird. It really looks like `tact?` sends back ` C` for the unit and `tset?` ` Celsius`. I guess we can add a replace routine that acts on Celsius before acting on C, just to be on the safe side in case this is a settings thing...

scasagrande, 2022-03-03:
Fixed in #333

closed 2022-03-03 by scasagrande