instrumentkit/InstrumentKit #332: usbtmc comunication

njrebelo, 2022-02-21:
I have the VID and the PID of the equipment, its a PM100A, but can't communicate, with open_usb or open_usbtmc.

[image]

if I want to try VISA how do I find all the information I need?

njrebelo, 2022-02-21:
I tried visa, but this happened:
[image]

But if I directly use the USBTMC library it works:
[image]

njrebelo, 2022-02-21:
So now I manage to comunicate, but get an error when I try to read: 
[image]

trappitsch, 2022-02-21:
Might be that this is related to #260. We had a bug fix in there that should take some weird behavior out from this class. Could you try the latest version of `instrumentkit`? You can install it via:
```
pip install -U instrumentkit --pre
```
That should install `v1.0.0.dev0`, where this this is hopepfully fixed.

I'm not fully clear on why it doesn't work with `open_visa`, that would be the preferred method. Maybe worth trying again via visa after updating `ik`.

njrebelo, 2022-02-28:
[image]
Used PyVisa commandline to find the correct VISA code and now it worked perfectly

scasagrande, 2022-02-28:
perfect!

closed 2022-02-28 by scasagrande