instrumentkit/InstrumentKit #347: open_gpibethernet()/GPIBCommunicator attribute error

pstrk, 2022-04-11:
I have been experimenting with the Prologix GPIB-Ethernet adapter and have been facing some issues to get it working.

The open_gpibethernet() method  in the instrument module calls the GPIBCommunicator() class with a socket object as the filelike argument:

```
conn = socket.socket()
conn.connect((host, port))
return cls(GPIBCommunicator(conn, gpib_address, model))
```
where conn is a socket object.

GPIBCommunciator then raises an AttributeError as the attribute 'terminator' is not defined for the socket object. I think the correct way is to provide a SocketCommunicator as the filelike argument instead. The following corrections seem to fix the issue for the Prologix adapter:

``` 
conn = socket.socket()
conn.connect((host, port))
comm = SocketCommunicator(conn)
return cls(GPIBCommunicator(comm, gpib_address, model))
```

scasagrande, 2022-04-11:
I believe you are correct

With those changes, does the GPIB-Ethernet adapter that you have, then function correctly?

pstrk, 2022-04-11:
My first tests show that I can communicate with the adapter after these changes, let me run a few more tests to check the communication with some actual equipment. Once this is done I can commit & push the changes and create a pull request.

pstrk, 2022-04-11:
Unfortunately it seems it does not solve all issues when communicating with equipment. I'll look into this in the coming days.

scasagrande, 2022-04-11:
No problem. I am interested in hearing more about this so please keep this updated with your findings :)

pstrk, 2022-04-11:
I have now a [working solution](https://github.com/pstrk/InstrumentKit/tree/feature_prologix_gpib_ethernet_support), but some cleanup is still required.

The main issue was that in the GPIB communicator class there was no clear distinction between the terminator for Ethernet commands (i.e. commands sent to the Ethernet adapter over the network) and the terminator for GPIB commands/in GPIB responses.

These two terminators might be different, e.g. when EOI is enabled there is no terminator for the GPIB commands, but the Ethernet terminator is still required. This lead to some major issues when reading back data from the instrument through the socket_communicator. In addition some required adapter commands were missing. The latest commit provides basic functionality with the Prologix GPIB-Ethernet adapter.