instrumentkit/InstrumentKit #350: PM100USB ThorLabs

njrebelo, 2022-05-09:
I am trying to set the wavelenght configuration from my PowerReads, but from your code I don't realaly now how to do it.
I've been doing it manually in the lab

scasagrande, 2022-05-09:
After opening connection, use the `PM100USB.read` function https://instrumentkit.readthedocs.io/en/latest/apiref/thorlabs.html#instruments.thorlabs.PM100USB.read

njrebelo, 2022-06-03:
Sorry to only reply now.
My problem is different. I am on the Power configuration, so the PowerMeter is giving me power reads in mW for a given wavelengths. I want to change that wavelenght through the code.
I can already read it.

Many thanks :)

njrebelo, 2022-06-03:
So I realized this specific to the PM100A, so I add a function to the library with the SCIP command that allows me to do this:

    def wavelenght(self, wav):
        self.sendcmd("CORR:WAV "+str(wav))

Thanks

closed 2022-06-03 by njrebelo