instrumentkit/InstrumentKit #354: Expected for unitful property setter without unit

bmoneke, 2022-05-12:
I noticed, that a unitful property setter does not raise an error, if a number is given. In this case, it just assumes the default unit.

For a bounded_unitful_property, the same command (inst.some_unitful_property = 500) raises an error, because the comparison in unitful_property._setter happens before assume_units is called.

What is the expected behavior? Should both raise an error or should both take the magnitude and assume the default unit?
I guess the second option is desired, because assume_units is called.

I'm confident to be able to fix the issue.

scasagrande, 2022-05-12:
Whoops! Yeah the desired behaviour for IK has been to assume the default units if none are supplied

scasagrande, 2022-05-12:
Accepting strings with units is an interesting idea. I'm not sure how I feel about accepting strings, and then querying the same property will return a numeric quantity. It feels wrong, but I do appreciate the idea of allowing people to set values without having to mess about with manually importing the unit registry

I can see the use case of a user just wanting to experiment with some equipment in their terminal. Passing in strings with units would be easier:

```python
inst.property = "123units"
```
vs
```python
from instruments.units import ureg as u
inst.property = 123 * u.units
```

trappitsch, 2022-05-12:
Just found this fitting comment this morning... good timing! 
https://www.nature.com/articles/d41586-022-01233-w

trappitsch, 2022-05-12:
> And they didn't reach out to us?? roll_eyes

I know... :shrug: 

With respect to reading in strings with units... Sounds like a nice idea, as long as `pint` handles everything I don't see too much of a downside. Meaning that the the following should all be equivalent - which is the case since `pint` handles it all:

```python
>>> from instruments.units import ureg as u
>>> a = u.Quantity(10, u.mm)
>>> b = 10 * u.mm
>>> c = u.Quantity("10mm")
>>> a == b == c
True
```

I see string parsing as the least favorable / important way of getting units in though, mostly because I rely on the fact that `pint` interprets `"10mm"` as millimeters. If I'm unsure about `u.mm`, I can always check the docstring, and then know it will work...
This argument might not make all that much sense, but when I use `ik`, I always want to load the unit registry and ensure that there's no ambiguity.

commit b6d54004 (2022-05-20): Fixes #354 bug that unitful_property with limits fails (#355)

* Fix bug that unitful_property with limits did fail on comparison with unitless value. Fixes #354.

Assume_units accepts strings as well.

* [pre-commit.ci] auto fixes from pre-commit.com hooks

for more information, see https://pre-commit.ci

* Unitful_property bounds use assume unit on bounds as well.

* [pre-commit.ci] auto fixes from pre-commit.com hooks

for more information, see https://pre-commit.ci

* Test adjusted, that limits functions return quantity.
closed 2022-05-20 by scasagrande