instrumentkit/InstrumentKit #359: Can't query for instruments using visa i.e. created with open_visa() method

JerPatton, 2022-06-01:
Hi all!

I am connecting to an Agilent 33220A via TCPIP and VISA using either of the below options:

`import instruments as ik`    
`inst = ik.agilent.Agilent33220a.open_visa("TCPIP0::127.0.0.1::inst0::INSTR")`
OR
`import instruments as ik`    
`inst = ik.agilent.Agilent33220a.open_from_uri("visa://TCPIP0::127.0.0.1::inst0::INSTR")`

The connexion works and I can send (write) commands to the device, BUT I can't query anything as I systematically get the following error:
`VisaIOError: VI_ERROR_TMO (-1073807339): Timeout expired before operation completed.`

After investigating, I found the closed issue #216  mentioning the exact same problem and I can reproduce the same behaviour with the command terminator. 
Atm, the terminator obtained via `inst.terminator` is `'\n'`, as expected for this device. Yet, setting the command terminator to `''` via `inst.terminator = ''` solves the query problems.

To me this behavior is curious. It looks like that now the normal command terminator `'\n'` is appended twice somewhere to the end of the SCPI command, which causes the error. Setting it to `''` removes the redundant one, but this is both counterintuitive and a bit inconvenient to set everytime a connection is made.

Currently I came up with two "quick" fixes: 
- Setting the terminator to `''` inside the VisaCommunicator class --> This may not be the best as it may break the communication for other devices that use another terminator.
- Adding an __init__() constructor inside the Agilent33220a class and overwriting the default terminator there (If all agilent have the same expected command an Agilent class could also be created for that purpose):
  ```
    def __init__(self, filelike):
        super().__init__(filelike)
        self.terminator = ''
  ```
What are your thoughts on the matter? Is that a bug or did I miss something?

trappitsch, 2022-06-01:
This is pretty strange. According to the [manual](https://www.keysight.com/us/en/assets/9018-04437/user-manuals/9018-04437.pdf), page 269, the device should send a newline `\n` terminator. 
If you set the terminator to `''`, can you still send commands without any issues? If not (which is my best guess), the instrument would require a termination command to send the command, but does not send one when sending a response. Then we could simply fix it in the read routine while leaving the send routine as is.

For your fixes: This is instrument specific, therefore the second solution would be the way to go. Otherwise, other instruments will surely break.

scasagrande, 2022-06-01:
Many moons ago I had no issues with any asymmetry in termination on Agilent instruments, but I also wasn't using VISA or TCPIP.

Can you check if the instrument itself for its termination character settings? Typically more modern instruments will have the ability to change these settings in its menus.

JerPatton, 2022-06-02:
Thanks for your replies!

@scasagrande the Agilent 33220A documentation [manual](https://www.keysight.com/us/en/assets/9018-04437/user-manuals/9018-04437.pdf) p. 279 specifies that  commands "must terminate  with a \<new line\> character" i.e. `\n`, or eventually "a \<carriage return\> followed by a \<new line\>" i.e. `\r\n`. There is also unfortunately no way of changing this setting manually or through SCPI commands for this device.

@trappitsch indeed that's more than strange... When I set the terminator to `''` I can write, read and query without error. To me, the fact that query commands work then is "normal", since a pyvisa `query()` implies a `write()` command followed by a `read()` command. 
Now, the more I dig into this issue, the more I am confused... ^^' 
I explain myself with a bit of code to look for what is sent via the different commands:

``` 
1) Current terminator: 
'\n'

2) write() + read() commands : inst.write('*IDN?') and then repr(inst.read())

--> Calls write_raw() of InstrumentKit VisaCommunicator: 
     with message: b'*IDN?'
--> Calls write_raw() of pyvisa MessageBasedResource:
     Mess: b'*IDN?'
Response from device: 'Agilent Technologies,33220A,MY44035935,2.07-2.02-22-2\n'

3) sendcmd() + read() command: inst.sendcmd('*IDN?') and then repr(inst.read())

--> Calls write_raw() of InstrumentKit VisaCommunicator: 
     with message: b'*IDN?\n'
--> Calls write_raw() of pyvisa MessageBasedResource:
     Mess: b'*IDN?\n'
Response from device: 'Agilent Technologies,33220A,MY44035935,2.07-2.02-22-2\n'

4) query() command:  repr(inst.query('*IDN?'))

--> Calls write() of pyvisa MessageBasedResource: 
     with message: '*IDN?\n\r\n'
--> Calls write_raw() of pyvisa MessageBasedResource:
     Mess: b'*IDN?\n\r\n'
---------------------------------------------------------------------------
VisaIOError                               Traceback (most recent call last)
Input In [1], in <cell line: 24>()
     21 print('\r')
     23 print('4) query() command:')
---> 24 print('Response from device: ' + repr(inst.query('*IDN?')))

File D:\Notebooks\venv_notebook\lib\site-packages\instruments\abstract_instruments\instrument.py:123, in Instrument.query(self, cmd, size)
    120     ack_expected_list = [ack_expected_list]
    122 if ack_expected_list[0] is None:  # Case no ACK
--> 123     value = self._file.query(cmd, size)
    124 else:  # Case with ACKs
    125     _ = self._file.query(cmd, size=0)  # Send the cmd, don't read

File D:\Notebooks\venv_notebook\lib\site-packages\instruments\abstract_instruments\comm\abstract_comm.py:232, in AbstractCommunicator.query(self, msg, size)
    230 if self.debug:
    231     self._logger.debug(" <- %s", repr(msg))
--> 232 resp = self._query(msg, size)
    233 if self.debug:
    234     self._logger.debug(" -> %s", repr(resp))

File D:\Notebooks\venv_notebook\lib\site-packages\instruments\abstract_instruments\comm\visa_communicator.py:184, in VisaCommunicator._query(self, msg, size)
    171 """
    172 This is the implementation of ``query`` for communicating with
    173 VISA connections. This function is in turn wrapped by the concrete
   (...)
    181 :rtype: `str`
    182 """
    183 msg += self._terminator
--> 184 return self._conn.query(msg)

File D:\Notebooks\venv_notebook\lib\site-packages\pyvisa\resources\messagebased.py:652, in MessageBasedResource.query(self, message, delay)
    649 if delay > 0.0:
    650     time.sleep(delay)
--> 652 return self.read()

[... 46 lines cut ...]

--> Calls write_raw() of InstrumentKit VisaCommunicator: 
     with message: b'*IDN?'
--> Calls write_raw() of pyvisa MessageBasedResource:
     Mess: b'*IDN?'
Response from device: 'Agilent Technologies,33220A,MY44035935,2.07-2.02-22-2\n'

3) sendcmd() + read() command: inst.sendcmd('*IDN?') and then repr(inst.read())

--> Calls write_raw() of InstrumentKit VisaCommunicator: 
     with message: b'*IDN?'
--> Calls write_raw() of pyvisa MessageBasedResource:
     Mess: b'*IDN?'
Response from device: 'Agilent Technologies,33220A,MY44035935,2.07-2.02-22-2\n'

4) query() command:  repr(inst.query('*IDN?'))

--> Calls write() of pyvisa MessageBasedResource: 
     with message: '*IDN?\r\n'
--> Calls write_raw() of pyvisa MessageBasedResource:
     Mess: b'*IDN?\r\n'
Response from device: 'Agilent Technologies,33220A,MY44035935,2.07-2.02-22-2\n'
```

However, what appears now is a clear asymmetry in command endings depending on the way we send the command... That is really curious behaviour.

trappitsch, 2022-06-02:
This is interesting, thanks for the details. It seems as if somewhere, a `\r\n` gets appended and we need to find where. This explains why the terminator must be set to `''` in order for it to work, otherwise you have a `\n\r\n` in the pipeline, it quits after the first `\n` and then there's still a `\r\n` in the buffer somewhere and all breaks...

I think this might have something to do with `pyvisa` itself. [Here](https://pyvisa.readthedocs.io/en/latest/introduction/resources.html#termination-characters) is a discussion on termination characters. From your above message, it looks like the `MessageBasedResource` class plays a role. Looking at the docs [here](https://pyvisa.readthedocs.io/en/latest/_modules/pyvisa/resources/messagebased.html#MessageBasedResource) shows that as the `_write_termination` character, `\r\n` is set by default. This is different from our `instrument.terminator` keyword and even though we inherit ultimately from the class, if we set the wrong property of the instrument in `ik`, then `pyvisa` has no way of knowing that we defined a termination character already. `ik` takes care of this character internally already, that would explain how you end up with a `\n\r\n` in the first case above and it would explain why things still work when you set `inst.terminator = ''`. 

When you initialize communcation, can you try and set `instrument.write_termination = ''` and leave the `ik` terminator as `\n`, then again try the query?

JerPatton, 2022-06-03:
Okay so I checked a bit further into these termination meanders.

So any connection to a device via TCPIP and VISA (PyVISA) with the following commands:
```
inst = ik.generic_scpi.SCPIFunctionGenerator.open_from_uri("visa://TCPIP0::127.0.0.1::inst0::INSTR")
inst = ik.agilent.Agilent33220a.open_from_uri("visa://TCPIP0::127.0.0.1::inst0::INSTR")
inst = ik.agilent.Agilent33220a.open_visa("TCPIP0::127.0.0.1::inst0::INSTR")
```
create a VisaCommunicator "file-like" object (accessed via `inst._file`), which in turns possess a pyvisa MessageBasedResource attribute for communication (accessed via `inst._file._conn`).

The nightmare starts here, as each level of these classes have/redefine their own default terminator. So the current (default) behavior, which causes an error for `query()` commands, is:
```
inst.terminator: '\n' 
Type(inst): <class 'instruments.agilent.agilent33220a.Agilent33220a'>

inst._file.terminator: '\n' 
Type(inst._file): <class 'instruments.abstract_instruments.comm.visa_communicator.VisaCommunicator'>

inst._file._conn.write_termination: '\r\n' 
Type(inst._file._conn): <class 'pyvisa.resources.tcpip.TCPIPInstrument'>  (This is just a wrapper for MessageBasedResource TCPIP/INSTR connections)
```
and the message sent by the `query('*IDN?')` is `'*IDN?\n\r\n'`

Now if try your suggestion @trappitsch, i.e. trying to set `instrument.write_termination = ''` at communication initialization, the only way I found is to set `ins = pyvisa.ResourceManager().open_resource(resource_name, write_termination='')` in `instrument.open_visa()` function.

Now the terminators are as follows and the `query()` commands succeed:
```
inst.terminator: '\n' 
Type(inst): <class 'instruments.agilent.agilent33220a.Agilent33220a'>

inst._file.terminator: '\n' 
Type(inst._file): <class 'instruments.abstract_instruments.comm.visa_communicator.VisaCommunicator'>

inst._file._conn.write_termination: '' 
Type(inst._file._conn): <class 'pyvisa.resources.tcpip.TCPIPInstrument'>
```
and the message sent by the `query('*IDN?')` is `'*IDN?\n'`.

--> So if I understand correctly, `pyvisa `already takes care of termination. 
Hence, for connections with a VISA ressource via `ik` would it be better to set the `write_termination` to `''` in `pyvisa.ResourceManager().open_resource()` **by default** and thus let the "control" of command terminators to `ik` **OR** set the default terminator to `''` somewhere in `ik` when creating a visa connection via the `instrument.open_visa()` method e.g. in the `VisaCommunicator` class (since `pyvisa` seems to takes care of that itself)?

trappitsch, 2022-06-03:
I got an oscilloscope standing around (in lab) that I can test the `open_visa()` method with as well, will give this a try next week too to check what is actually sent to the device... Just to double check :)

trappitsch, 2022-06-06:
Tested the `open_visa` with the Teledyne LeCroy scope and indeed, the command that is sent in the query is also: `\n\r\n`, the first `\n` coming from the terminator. Seems to work, however, it's not what we want as the behavior in `ik` I think. I can easily see how this might throw issues. @scasagrande: what do you think on how to proceed?

JerPatton, 2022-06-07:
Thank you for further investigating on this issue! 

Now that I understand better I can tune it as I want for a quick fix, but I am really interested in your thoughts and decisions on the matter. If I can be of any help, please let me know too.

JerPatton, 2022-06-14:
@trappitsch thanks for your input. 
I don't know if this was adressed to me, but IMO it sounds like a good option. It would remove the duplicated terminator setting when communicating through VISA, while still leaving the ability for changing it if desired.
Let me know if you need some testing for that.

scasagrande, 2022-06-14:
@trappitsch I think that is reasonable. That should also cover the potential future case of, for example, having another connection wrap a VISA connection. Maybe some sort of SSH connection. I don't know, just throwing out possibilities there. Similar to how the GPIB connection is wrapped by a serial, or ethernet, connection.

trappitsch, 2022-06-15:
An implementation of this is now in PR #362. @JerPatton: Would be great if you could test this. You can simply check out this branch and give it a try:
https://github.com/trappitsch/InstrumentKit/tree/fix_terminator_visa_comm

closed 2022-06-15 by scasagrande

JerPatton, 2022-06-20:
@trappitsch super nice! I'll test it on my side just to be sure and I'll let you know. But I believe everything should be fine now. 
Thank you both for your reactivity!

PR #362 (2022-06-15): Fix terminator issues with pyvisa
As lined out in #359, here are some fixes for the `pyisa` terminator.

`pyvisa` uses `read_termination` and `write_termination` props to set the respective terminators for commands ([see here](https://pyvisa.readthedocs.io/en/latest/introduction/communication.html?highlight=read_termination#getting-the-instrument-configuration-right)). This has led to issues in `ik`, which uses by default a `terminator` property to set terminator for all instruments. The `VisaCommunicator` class was extended to:

- Expose `read_termination` and `write_termination` properties t [... cut]