instrumentkit/InstrumentKit #368: error working with SC10

Mendelevium9555, 2022-09-13:
I have been experimenting with the SC10 shutter controller and have been facing a issue getting it to work.

`import instruments as ik
sc = ik.thorlabs.SC10.open_serial("COM6", 9600, timeout=100)
print(sc.name)
print(sc.enable)`

After I call any sc.XXXX command at first time, everything works fine. But the code will fail if I make a second command, showing

`instruments.errors.AcknowledgementError: Incorrect ACK message received`

trappitsch, 2022-09-14:
What version of `instrumentkit` are you using? The error reminds me a lot of #296, which was fixed in #303. If you are using the latest stable release from pypi, `v0.6.0`, please install the `v1.0.0b1` pre-release. I think that should fix it. To install the pre-release:

``
pip install instrumentkit --pre
``

scasagrande, 2022-09-14:
Yes can you report what version you are using?

Mendelevium9555, 2022-09-14:
Sorry for the late reply. I don't know how to check the version. What is the command I can check the version of instrumentkit. I just used `pip install instrumentkit` to do the installation. Additionally, I am confused about how to enable the shutter to open. By looking at the code, I feel like it should be `sc.enable = True`, but I will get the error from sendcmd.

Mendelevium9555, 2022-09-15:
I have updated the version to `v1.0.0b1` and the issue has been solved. Thanks for your help. However, I still have a question on how to set the shutter to be enabled. I have also tried `sc.sendcmd("ens=1")` but it did not work as well.

trappitsch, 2022-09-16:
Two things that come to mind to check:
1. Did you check what mode your shutter is in? In order to see it open / close depending on what you send to enabled, you should make sure that it's (for testing purposes) set to manual. Something like this should work, assuming `inst` is the opened instrument:         
   ```python
   inst.mode = inst.Mode.manual
   inst.enable = True
   ```
   I hope the shutter opens then. If this works, could you try to open the shutter twice in a row and see if it stays open? Would be good to see how it behaves, see below.
2.  If this does not work, then the [manual](https://www.thorlabs.de/drawings/d41a757f77ef33f7-B2EB5952-F27C-6800-E2D3774C5CC275F9/SC10-Manual.pdf) might actually be correct (not always the case). The manual states that the shutter is simply toggled by sending `ens`, without any arguments. However, `ik` currently uses a `bool_property` to enable / disable the shutter, which means it sends `ens=1` to open or `ens=0` to close. Assuming 1 worked, the last comment would check if it simply toggles (and ignores the `=1`, or if it actually reads the `=1` and sets the property accordingly. If (1) above failed, can you, after setting the shutter to manual mode, send the following command: ``inst.sendcmd("ens")``? If the shutter then opens (but not in 1), there is indeed still a bug in this class, but should be easily fixable.

I hope one of these points work, don't have a good  number 3 in mind right now. Let us know if this helps and what you find. Thanks!

Mendelevium9555, 2022-09-18:
Finally, I used 
```
inst.mode = inst.Mode.manual
inst.sendcmd("ens")
```
and it works. Thanks for your support. I think there might be a bug in `bool_property'.

trappitsch, 2022-09-20:
Thanks for checking. Looks like there is an error in the `bool_property`. I propose the following:
- We rewrite the `inst.enable` routine to not use a bool property as it is currently, but rather implement it by itself.
- Property read should return the status as per manual
- Property write: First read the current status, then send the status requested by the user (if necessary): This would mean that `inst.enable = True` would still open the shutter and `inst.enable = False` would still close it, but a command is only sent if the status is actually the opposite of what the user requests. 

I think that this solution would remain in the `ik` spirit. Thoughts @scasagrande?

scasagrande, 2022-09-20:
Yeah, if the device is expecting `ens` to simply toggle the state, then your suggestion is correct @trappitsch 

We've had to do something similar to other instruments, where the commands aren't idempotent

trappitsch, 2022-09-20:
From @Mendelevium9555 report and the manual, it seems to me as if indeed it is a toggle and not a setting. I rewrote the `enable` command as described above, see PR #370

trappitsch, 2022-09-21:
@Mendelevium9555: This should now be fixed. Would be great if you could install the latest version directly from GitHub with 

```
pip install -U git+https://www.github.com/instrumentkit/InstrumentKit.git
```
You should then be available to run the shutter as you would expect, i.e., with `sc.enable = True` or `sc.enable = False`. Let us know if it works or not, since you have the hardware. Thanks!