instrumentkit/InstrumentKit #375: test_thorlabs::test_abstract::test_query_packet_no_response_with_timeout gives flakey results

Gracecr, 2022-10-07:
On my system, [test_query_packet_no_response_with_timeout](https://github.com/instrumentkit/InstrumentKit/blob/main/tests/test_thorlabs/test_abstract.py#L132) frequently fails.

OS: Window 10
Python versions: 3.6, 3.7, 3.8, 3.9, 3.10

I generally run my tests using VSCode's test runner, though the issue still intermittently appears when running on the command line with `tox`. If I attempt to debug the test, calls to `time.time()` continue to increase. I'm guessing the VSCode debugger is making these calls.

It seems like relying on nothing else calling `time.time()` is a bit fragile, at least in my case. Is the check:

`assert inst.querypacket(example_packet, timeout=0) is None`

sufficient to show that the function is working as expected?

trappitsch, 2022-10-28:
Testing is currently being looked over in #374, so this might be an issue that could be tackled in that specific or an associated PR. Are the flakey tests associated ones that use `hypothesis` by any chance? 

I'm not sure about the `time.time()` calls. Certain instruments have `time` included with sleeps in order to make some flakey communication more stable. This is generally based on experience values with the hardware, should also be user settable however. Furthermore, in some / most / all (?) of the tests, the time calls should be mocked... no need to wait when we test. Not sure this addresses any part of your issue actually though :)

Gracecr, 2022-10-31:
Don't think it's related to #374 as this particular test doesn't use hypothesis.

My machine is executing [the loop in ThorLabsInstrument.query_packet](https://github.com/instrumentkit/InstrumentKit/blob/main/src/instruments/thorlabs/_abstract.py#L77-L89) multiple times before `time.time()` has a chance to update, so the test fails as `time_spy.call_count > 2`. Adding a `time.sleep(1e-32)` above line 87 fixes the issue for me. I put up #379 for it, though it is a hacky solution.

trappitsch, 2022-10-31:
Ah, you need a slower machine ;) I don't like the fix, since it fixes a flaky test in the driver itself. We should fix the test though.