instrumentkit/InstrumentKit #401: How to capture live running traces from MSO7104B oscilloscope with the help of MATLAB without any delay?

imgb27, 2023-07-16:


trappitsch, 2023-07-17:
Not excatly sure what the issue is / what you would like to exactly do from the title alone, some description would be useful. 
However, a brief search of MSO7104B shows me that this is a Keysight instrument, might have been a former Agilent instrument, which doesn't seem to be a part of `ik`...

imgb27, 2023-07-17:
Sir, I'm using MSO7104B Agilent Oscilloscope for capturing traces. I'm using MATLAB for storing traces. Connection is established between both the devices with VISA interface. Now, i want to capture the continuous running live traces from the oscilloscope and store it in MATLAB for further analysis. Can you guide me in sorting out this issue?

trappitsch, 2023-07-19:
Well, I can't really give you any detailed ideas since (1) `instrumentkit` does not have that instrument in the database and (2) it's a python and not a Matlab driver (although, there seems to be some way to do python drivers in Matlab). 

I guess you'll need to read the necessary matlab documentation on how to communicate and then read the programmers manual for the instrument in order to know what to send / receive. Programmers guide seems to be [here](https://www.keysight.com/us/en/support/MSO7104B/mixed-signal-oscilloscope-1ghz-4-analog-16-digital-channels.html). Hope this helps.

imgb27, 2023-07-28:
Thanks for replying mate. But anyways the issue is still on.

scasagrande, 2023-09-29:
Not much we can do here 🤷‍♂️

closed 2023-09-29 by scasagrande