instrumentkit/InstrumentKit #406: Dependency update

trappitsch, 2024-01-28:
With moving to support 3.12 (PR #405) it might be time to look at some pinned dependencies again. Currently, the following are pinned:

```
pint>=0.16.1,<0.21.0
ruamel.yaml>=0.16,<0.17
```

- `ruamel.yaml`, see PR #327, should be a fairly easy fix.
- `pint` was pinned in PR #396. Since we are dropping py37 (EOL), this warrants revisiting.

On the dev side, there are:

```
hypothesis~=6.56.0
pytest~=7.2.0
```

Both of these were pinned in #374 

- `pytest v8` is out and should be. Since they tend to not have many breaking changes even with major version upgrades, this should hopefully not be too difficult.
- `hypothesis` was pinned and some flaky tests fixed.

For both sides, I would propose to pin only minimum versions and leave upper versions open. Especially since the `pre-commit` CI runs regularly to update the hooks and then runs all the tests again, we should be able to catch any bugs that occur from newer versions very fast. This strategy has worked pretty well for some of my other projects. 

Finally and with respect to tests: We should also get rid of the depreciation warnings that do/might show up. Just a good precaution for future updates. 

Any comments welcome, I picture we'll break this into a few individual PRs just to make sure that things are working at every step and we can - if necessary - incrementally go back. My idea was one PR per main dependency and one for the whole testing/depreciation business. What do you think @scasagrande?

trappitsch, 2024-01-29:
Update on `pint`: They dropped support for Python `3.8` as of version `0.22`. Since we should support `3.8` until EOL, I upgraded the pinning for now to `~=0.21`, see #408.

scasagrande, 2024-01-29:
> For both sides, I would propose to pin only minimum versions and leave upper versions open.

I actually don't like this for development tools. I've seen far too many bugs and errors both in this project and in my professional projects due to floating development dependencies.

trappitsch, 2024-01-29:
That's a good point. I'll have a look at updating to a more recent version to keep it from falling behind on the dev side. Regular incremental upgrades are still useful I guess.

PR #411 (2024-02-07): Update dev dependencies
Ok, last one for #406:
Updating `pytest` and `hypothesis` to the latest versions. Also added a missing `__init__.py` file, which `pytest` complained about is missing due to the change in `8.0.0` of test collection.

As discussed in #406, these dependencies are still pinned to minor versions, however, to new ones. Should keep us future proof for a while longer.

Looks like all passing and good. Closes #406
closed 2024-02-07 by scasagrande