instrumentkit/InstrumentKit #427: Support for python 3.13, drop 3.8 as EOL

trappitsch, 2024-11-04:
It's this time of the year again: python 3.13 is out and 3.8 is officially EOL. Currently the tests dont succeed yet under 3.13. Waiting for pint to fix an issue, see [here](https://github.com/hgrecco/pint/issues/2065). Keeping an eye on this and will test again.

trappitsch, 2024-11-12:
Well, pint seems to be fixed at this point, however, there is one more larger issue: `python-vxi11`. It depends on `xdrlib` (was part of the standard library), which was deprecated in 3.11 and removed in 3.13. A summary of the issues and a possible fix is discussed here: https://github.com/python-ivi/python-vxi11/issues/49

Unfortunately, `python-vxi11` seems to be not maintained anymore. This is a larger problem for `ik`, I'll put more thoughts into a new issue.

scasagrande, 2025-02-13:
3.13 now supported!

closed 2025-02-13 by scasagrande