instrumentkit/InstrumentKit #429: python-vxi11: incompatible with python 3.13 and likely unmaintained

trappitsch, 2024-11-12:
As mentioned in #427, `python-vxi11`, which `ik` depends on, is broken in python 3.13 because it depends on `xdrlib`, which has been removed in 3.13 from the standard library.
Furthermore, `python-vxi11` also seems to be not maintained anymore. Last commit was ~7 years ago and 10 open PRs that proposed various fixes over the years for issues that have come up, i.e., [an `xdrlib` replacement fix](https://github.com/python-ivi/python-vxi11/issues/49). Contacting the author has been tried some years ago, but not resulted in anything ([see here](https://github.com/python-ivi/python-vxi11/issues/38)). 

*Note for the future:*
Looking through our dependencies, this will most likely happen sooner or later as well with `python-usbtmc`, which comes from the same author(s) as `python-vxi11`. Similar story here, no commits in ~7 years and 9 open PRs, however, it doesn't seem broken at this point under 3.13.  

**What to do? **

A quick search on pypi revealed no clear alternative solution for our issue. Since we depend on `python-vxi11` and probably don't want to drop support, I think it would be reasonable to discuss maintaining a fork of the project in the `ik` organization. `python-vxi11` is MIT licensed, so this should all be fine. A first push, just to get it compatible without errors and warnings would be:

- Create a maintained fork under a new name (that can also be released on `pypi`)
- Add `xdrlib3` as a dependency (the first one) to replace the removed `xdrlib`: I tested this, it's plug and play.
- Fix the regex warnings.
- Drop `nose` and add `pytest` in order to get the (very sparse) test suite to pass.
- Cleanup and modernization of repo
  - Docs - go through, clean up, put on rtd
  - Installation - turn into `pyproject.toml` based project (easy, as there's going to be basically one dependency).

Down the line and in order to maintain it easily, we'd have to add a better test suite and even might want to declare certain parts (e.g., the CLI interface) as unmaintained... 

I think it would make some sense to put this in the `ik` organization, as there is already a community around `ik`, which would make it more sustainable in the future. I'd be happy to do a push for implementing these first fixes / modernizations and to help with maintaining, but let's discuss this in more detail first.

scasagrande, 2024-11-13:
This is actually very good for me to know, because I found something at work that depends on python-vxi11 😂 

I think you're right, it should be easy for us to take on a fork. `py-vxi11`? what are we thinking

scasagrande, 2024-11-13:
i'll try to email the author and see if they have any plans

http://www.alexforencich.com/wiki/en/start
https://github.com/alexforencich

trappitsch, 2024-11-13:
Sounds great, thanks! Yes, an e-mail would for sure be a good and nice thing to do first. If there's no answer, `py-vxi11` sounds great - it's even shorter :)

trappitsch, 2024-12-13:
Any update here @scasagrande?

scasagrande, 2024-12-16:
I actually forgot to contact the author, whoops! However in the mean time can we try the temporary migration fix mentioned here: https://github.com/python-ivi/python-vxi11/issues/49#issuecomment-2534310413