instrumentkit/InstrumentKit #430: virus detected

m-utkur, 2024-12-02:
I was trying to download this package but Chrome says virus detected. Any idea why? Thanks.

scasagrande, 2024-12-02:
No idea. I just tried on chrome and I can reproduce the error. The tar.gz
version downloads just fine. I uploaded it to virustotal just in case but
it's all clean.

There's not much to the repo so I'd need a more detailed report to
understand why it's being flagged. Maybe because there is a PDF? It was
added ages ago direct from the manufacturer, but maybe it's making chrome
freak out.

trappitsch, 2024-12-03:
Weird indeed. Assuming you mean the Thorlabs APT rev 9 pdf... that could easily be removed, as it is pretty outdated (the version in ik is rev 9, the [most current version](https://www.thorlabs.com/software_pages/ViewSoftwarePage.cfm?Code=Motion_Control&viewtab=1) rev 39). 
I don't see any other thing that might trigger, but might be worth a try as it is also cleaning up.

That being sad and in a similar spirit, I just saw [this post](https://blog.pypi.org/posts/2024-11-14-pypi-now-supports-digital-attestations/) on digital attestation on `pypi`. This is probably of interest for a future release.

scasagrande, 2024-12-16:
I've read that browsers can automatically flag downloads as viruses if you're downloading a zip file and it contains certain file types within it. Maybe PDF is one that automatically triggers this?

PR #433 (2025-01-15): Remove Thorlabs APT Manual (pdf) from source code
Removing the very old and outdated Rev 9 of the Thorlabs APT Manual from the source code repo, in the hope that this fixes #430. 

The current/latest version of the Thorlabs motion controller communication protocol can generally be found on the [Thorlabs website](https://www.thorlabs.com/software_pages/ViewSoftwarePage.cfm?Code=Motion_Control&viewtab=1).
closed 2025-01-15 by scasagrande